librtlsdr/librtlsdr #16: rtl_tcp with Direct Sampling mode for V3 dongles.

racerxdl, 2016-08-31:
@ja450n implemented a call in rtl_tcp to use direct sampling. We should get his patches and apply to our development branch.

https://github.com/ja450n/librtlsdr

hayguen, 2016-08-31:
Yes, looks simple and good.
For rtl_tcp, i don't really see the need, as you can select/command that through the plugins of SDR# and HDSDR on Windows.
Probably some other apps (under Linux) don't support/control that ?!

More important, from my point of view is that option for rtl_fm and rtl_power. But here, some extra glue is needed: depending on frequency, automatic switching from direct sampling to I/Q sampling.
Below a frequency threshold, usually where the tuner is not capable to tune, direct sampling is to be used.
When tuner is able to tune to a frequency, then quadrature (I/Q) sampling should be used.

I'd tend do that switch automatically - per frequency tune - inside the librtlsdr!
That way we would just need a function to inform librtlsdr to auto-enable direct sampling. And it uses direct sampling, whenever frequency is too low for the tuner.

Hmmmm. In combination with bias-T ... probably that one should be switched off automatically on low(er) frequencies?! Or auto-activate bias-T only in a given/parametrized frequency range?
How does a special band preamp, e.g. ADS-B, behave on other frequencies when bias-T on / off?

One application: use bias-T for switching antenna with a relais.

What do you think?

racerxdl, 2016-08-31:
About the rtl_tcp, that depends how you use it. If you use with some frontend software thats fine. But some ppl use directly configured parameters and just get the stream output. In this case it is important to be able to enable direct_sampling by args.

Thats hard to assume, 820T2 can go down to 24MHz, but that does not mean that the user wants to use the tuner. I'm not a HF band reader, I think probably we need to ask to ppl that use the HF Band to see what they want.

hayguen, 2016-08-31:
I'm more developer than listener and listening HF from time to time.
However, i don't see a reason for using direct sampling where the tuner could be used, because the tuner has at least the IF filtering and and reduces the dynamic range for the ADC.
In addition you can use the tuner's LNA, Mixer and VGA.
But you never know ... if there really should be some application, i'd be interested to know!

racerxdl, 2016-08-31:
Yeah makes sense. I will see if I can find HF users to make opinion about.

hayguen, 2016-09-01:
Have a look at my last commit in the development branch:
it uses same option -D for the direct sampling mode, when arg is less than 5. Else it is interpreted as threshold frequency, below which in modes 3 and 4 direct sampling is activated.
Using modes 3 / 4 with threshold 0, librtlsdr should use the minimum tunable frequency itself. But i don't know for most tuners!

denhamgithub, 2017-01-12:
As a lower-power and 'cheap' ham radio operator, the idea of declaring something like -D for direct sampling mode in V3.0 would be deeply appreciated. Despite other comments. I like to listen to HF broadcasting, Radio Australia, and the like but carrying my ham radio equipment into the front room just cannot happen due to disability. I don't know whether the author of Linux GQRX supports this issue but he sounds pretty busy and perhaps this would help him add to his GUI? SDR# is Windows and I prefer Linux.

racerxdl, 2017-01-15:
@denhamgithub the author of GQRX is @csete but I think GQRX already supports that, since it supports any rtlsdr parameters in the command line.

denhamgithub, 2017-01-15:
Thank you Lucas – – I just found a way to go to HF according to a post I did not know from the author.

 It had to do with running the GUI , clicking on I/O and then entering an additional string into the ID. So I just added a comma to that plus direct_samp=2. Before I entered that string , lowest bandwidth shown was 24 MHz. After entering that string, and waiting a bit, then I could put in HF frequency like WWV 10.000 and hear the time station. Plus other HF frequencies. Reception is nowhere as good as my Elecraft K-3 amateur radio transceiver but pretty decent nonetheless.

pumatrax, 2017-02-23:
can this patch be done to rtl_sdr program as well. I would like both since I need to enable Q branch when I use OpenWebRX. There is a fork for rtl_sdr which works but the option is not there for rtl_tcp.

hayguen, 2020-08-18:
just added driver option '-O' for direct sampling mode 'dm=' and threshold frequency in the development branch.
all programs using the rtlsdr_set_opt_string(), most of rtlsdr tools do, will have handled the direct sampling mode in the driver.

example call:
```rtl_sdr -O dm=3:dm=20E6 -f 100.5M - >/dev/null```

usage part:
```
$ rtl_sdr
rtl_sdr, an I/Q recorder for RTL2832 based receivers
rtl_sdr version 0.7 github.com/hayguen (Aug 18 2020)
librtlsdr version 0.7 github.com/hayguen (Aug 18 2020)

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
...
	[-O	set RTL driver options seperated with ':', e.g. -O 'bc=30000:agc=0' ]
		f=<freqHz>            set tuner frequency
...
		ds=<direct_sampling>  deactivate/bypass tuner with 1
		dm=<ds_mode_thresh>   set dynamic direct threshold mode or threshold frequency:
		                        0: use I & Q; 1: use I; 2: use Q; 3: use I below threshold frequency;
		                        4: use Q below threshold frequency (=RTL-SDR v3)
		                        other values set the threshold frequency
		T=<bias_tee>          1 activates power at antenna one some dongles, e.g. rtl-sdr.com's V3
```

closed 2020-09-03 by hayguen