librtlsdr/librtlsdr #17: Some improvements from another branch

racerxdl, 2016-08-31:
@keenerd made some interesting changes in rtl_power and rtl_fm. We should take a look carefully at his changes and maybe merge to our branch. 

https://github.com/keenerd/rtl-sdr/

hayguen, 2016-09-03:
the patches i saw look very interesting.

ddcc, 2016-10-07:
Yes, it looks like he made a number of fixes and improvements, especially to `rtl_power`. I've been trying to track down a NaN output bug on ARM 32-bit, and it looks like it was fixed a while back in his branch: https://github.com/keenerd/rtl-sdr/commit/1b1c03d4e9330c30c9f4cc6de11fbe1572664fe7

racerxdl, 2016-10-07:
@ddcc this changes makes a lot of sense. Since long size is not standard across platforms int64_t would be a best alternative. This weekend I will take a look in his changes and make a list of stuff that we need to port. Then anyone can pick a task to port something.

pumatrax, 2017-02-23:
Yes please he has the sample mod in rtl_sdr

hayguen, 2020-08-19:
cherry picked that commit .. and fixed merge conflicts  into the development branch.
please have a look/try!

ValZapod, 2020-08-24:
https://github.com/keenerd/rtl-sdr/commit/a2d23c590ff97821127d04c18009fab16f964d6f
I adpated patch for our case

```
diff --git a/src/rtl_fm.c b/src/rtl_fm.c
index abd7977..c092e5d 100644
--- a/src/rtl_fm.c
+++ b/src/rtl_fm.c
@@ -1316,9 +1316,9 @@ void full_demod(struct demod_state *d)
 
 static void rtlsdr_callback(unsigned char *buf, uint32_t len, void *ctx)
 {
-       struct dongle_state *s = ctx;
-       struct demod_state *d = s->demod_target;
-       struct cmd_state *c = d->cmd;
+       struct dongle_state *s;
+       struct demod_state *d;
+       struct cmd_state *c;
        int i, muteLen = s->mute;
        unsigned char sampleMax;
        uint32_t sampleP, samplePowSum = 0.0;
@@ -1329,6 +1329,9 @@ static void rtlsdr_callback(unsigned char *buf, uint32_t len, void *ctx)
                return;}
        if (!ctx) {
                return;}
+       s = ctx;
+       d = s->demod_target;
+       c = d->cmd;
        time(&rawtime);
        if (duration > 0 && rawtime >= stop_time) {
                do_exit = 1;

```

ValZapod, 2020-08-24:
Wait a second, there is PPM calibration value in eeprom? Really??? https://github.com/keenerd/rtl-sdr/commit/389de6d8ed7e6c34e570bbac04c6007b4a19cc34

hayguen, 2020-08-24:
> keenerd/rtl-sdr@a2d23c5    I adapted patch for our case

i prefer NOT to use this. this would mean libusb / librtlsdr is really broken and erroneous in passing the context pointer.
if this fails .. it will fail forever .. and IMHO it's better to see this problem with a crash!
despite that, i like the c++ style of directly initialize variables ..

hayguen, 2020-08-24:
> Wait a second, there is PPM calibration value in eeprom? Really??? keenerd/rtl-sdr@389de6d

sounds cool for cheap dongles without TCXO :-)
i would prefer putting evaluation of serial number (for ppm value) and call to rtlsdr_set_freq_correction() directly into rtlsdr_open().
this way, one would not need to modify all the tools .. ?

ValZapod, 2020-08-24:
> if this fails .. it will fail forever .. and IMHO it's better to see this problem with a crash!

It will **not** fail. Technically -> is allowed to dereference null pointer as it is **ACCESSING** the data on null pointer that is illegal, here there is no access. It will use offsetof macro instead in some cases in others there will be just no effect. It is a very crazy topic, and better to just not use it. https://stackoverflow.com/questions/28482809/c-access-static-members-using-null-pointer/28483477 There is also issue in standard http://www.open-std.org/jtc1/sc22/wg21/docs/cwg_active.html#232

ValZapod, 2020-08-24:
> this way, one would not need to modify all the tools ..

Is that a problem? You do not need all that reading in rtlsdr_open().

hayguen, 2020-08-24:
> Is that a problem? You do not need all that reading in rtlsdr_open().

sorry, i don't understand your answer. can you explain?
what i meant, is not just my laziness in modification of tools, but you should also think of 4rd party tools, e.g. kal

ValZapod, 2020-08-24:
> 4rd party tools, e.g. kal

Will not it broke them then? Also non-GPL programs need to use rtl_tcp.c, right?

ValZapod, 2020-08-24:
> 'automatic' frequency correction with

Oh. So it is not really automatic? Is not done by hardware, i.e.? And also this is just a private format to store ppm value in serial number? Okay. Maybe it is not that interesting then.

hayguen, 2020-08-24:
from what i see in the sources of the patch:

> So it is not really automatic?

not full automatic

> Is not done by hardware, i.e.?

just in sense of reading/writing to EEPROM.
but you still need to:
* run rtl_test, note result
* run rtl_eeprom to modify serial number with correctly formatted string containing the ppm value, e.g. "A 23p", for +23ppm A being serial "_number_". note: named serial "_number_" .. you might use non-digit characters

> And also this is just a private format to store ppm value in serial number?

yes, sort of 'misuse' of serial number:
have a serial code/string followed by a space, the ppm value and 'p'.

ValZapod, 2020-08-24:
> sort of 'misuse' of serial number:

Okay, I do not like it, especially considering this: https://github.com/steve-m/librtlsdr/issues/33

So, another topic, is this possible bug? https://github.com/steve-m/librtlsdr/issues/12

hayguen, 2020-08-24:
> especially considering this: steve-m#33

think, that in the end, it leads to this accepted PR: https://github.com/librtlsdr/librtlsdr/pull/66
it would be nice to have a good/automatic test procedure to create a plot as in for https://github.com/steve-m/librtlsdr/pull/10
there must be a test/detection program for detecting the frequency of this GSM 954.0 MHz FCCH burst!? does kal use that?

> is this possible bug? steve-m#12

the referenced hint https://github.com/steve-m/librtlsdr/commit/9e23e728c30e19c0d1e4e0b32f199679be9c2ac1 is: downcasting an index to **device_count** . for a device count, even an 8 bit counter would be sufficient .. this can't be relevant

hayguen, 2020-08-24:
this thread is getting more than unreadable.
i would suggest creating new specific issues .. which might reference this one!

ValZapod, 2020-08-24:
I think it is all. All other changes, like register cachinb and rtl_power are too compicated.

hayguen, 2020-08-24:
thank you very much.

let me repeat the following - for new readers

> this thread is getting more than unreadable.
> i would suggest creating new specific issues .. which might reference this one!

closed 2020-09-03 by hayguen