librtlsdr/librtlsdr #26: What is the packet format of rtl_tcp?

chibill, 2016-11-28:
I am looking for any documentation on the format of packets used to communicate with rtl_tcp. Any idea?

racerxdl, 2016-11-28:
No packet format. Just raw I/Q Data.

chibill, 2016-11-28:
okay? so how do I send it commands like to tune to a different freq? and such. (I know it does that.)

And how do I use said I/Q data from a socket.

racerxdl, 2016-11-28:
Ok so the control packets are controlled here: 
https://github.com/librtlsdr/librtlsdr/blob/master/src/rtl_tcp.c#L297-L393

We don't have a documentation (for now) for this part. But its basically you will always send a struct like this:

```c
struct command{
	unsigned char cmd;
	unsigned int param;
}
```
Being `cmd` one of the values of this enum
```c
enum RTL_TCP_COMMANDS {
    SET_FREQUENCY             = 0x01,
    SET_SAMPLE_RATE           = 0x02,
    SET_GAIN_MODE             = 0x03,
    SET_GAIN                  = 0x04,
    SET_FREQUENCY_CORRECTION  = 0x05,
    SET_IF_STAGE              = 0x06,
    SET_TEST_MODE             = 0x07,
    SET_AGC_MODE              = 0x08,
    SET_DIRECT_SAMPLING       = 0x09,
    SET_OFFSET_TUNING         = 0x0A,
    SET_RTL_CRYSTAL           = 0x0B,
    SET_TUNER_CRYSTAL         = 0x0C,
    SET_TUNER_GAIN_BY_INDEX   = 0x0D,
    SET_TUNER_BANDWIDTH       = 0x0E,
    SET_BIAS_TEE              = 0x0F
};
```
that is described here: https://github.com/librtlsdr/librtlsdr/blob/master/include/rtl_tcp.h#L30-L46

The IQ data is raw 8 bit. You use the same way as the normal library. You will always receive 8 bit I/Q data interleaved ( I Q I Q I Q I Q ). Then what to do with this data is up to your program.

chibill, 2016-11-28:
Thanks.  ( working on a python client receiver lib)

closed 2016-11-28 by chibill