librtlsdr/librtlsdr #32: Most windows builds fail to run.

chibill, 2017-01-04:
They fail to run and say that two dll files are missing.

pthreadGCC2.dll
and
libgcc_s_dw2-1.dll

I am running windows 10 
Microsoft Windows [Version 10.0.14393]

chibill, 2017-01-04:
The libgcc_s_dw2-1.dll dependancy is because of minWG / gcc not being configure right. (According to what I have found while trying to fix this.)

racerxdl, 2017-01-04:
The pthreadGCC should also be included in the Mingw, but you can download it: https://sourceware.org/pthreads-win32/ 

The libgcc I dont get any problem here. What mingw are you using?

chibill, 2017-01-04:
none I downloaded the windows binaries from releases.

chibill, 2017-01-04:
for the pthreadGCC I was able to use the one from the old librtlsdr windows binary zip. (The one this fork is based off of.)

hayguen, 2017-01-05:
Try this release:
https://github.com/librtlsdr/librtlsdr/releases/tag/win32_with_biasTee

It's build statically (RTL_FULL_STATIC_BUILD), that you don't need any DLLs.

chibill, 2017-01-05:
Okay. I will.

chibill, 2017-01-05:
Now I have another problem that is directly related to this. I need the librtlsdr.dll for the one python lib I use. And I can't load it with this. (I never had this problem on my old laptop. Only once I got my new one.) I might need to install mingw to make this stuff work correctly.

racerxdl, 2017-01-06:
What is the issue you're having with the python lib? Not sure what python lib you're using, but it may need a specific compiled version of the python lib to work with upstream.

chibill, 2017-01-06:
It was just that the librtlsdr.dll has 4 dependacies on other dll files not shiped with it. (libusb.dll ,  libgcc_s_dw2-1.dll,  libwinpthread-1.dll and also pthreadGC2.dll) Most of which are shipped with mingw. 

So I had to pull those files off my old harddrive to get it to work. (As I got totally confused on how to reinstall mingw.) 

The lin in question was pyrtlsdr

racerxdl, 2017-01-06:
Actually it was my bad. I forgot to put all the deps in the packaged library for windows. I will update it tomorrow and put the needed deps.

chibill, 2017-01-06:
Okay. And it's fine. I was jsut so confused on why it worked on my old laptop and not my new one till I realized I have minwg installed on my old one

chibill, 2017-04-04:
Okay. Welp I am still sort of having this problem... But its more of do you have a 64 bit build? Because 64 bit python refuses to load the 32 bit version.

closed 2017-04-04 by chibill