librtlsdr/librtlsdr #34: Merge the R820t tuner fixes by @mutability

tomtastic, 2017-01-11:
Merge the R820t tuner fixes by @mutability from the pull request here : https://github.com/steve-m/librtlsdr/pull/10/commits/31ce37dc6448f747280e330b183e54ab6d8a544c

> Improve tuner precision by calculating the VCO divisor at full precision,
> not at kHz resolution. Also replace the manual divison loop with a
> simpler fixed-point calculation.

racerxdl, 2017-01-15:
👍 I will take a look on it.

Romeo-Golf, 2017-02-12:
Looks useful ;)

E3V3A, 2017-06-30:
Would this also fix the annoying `[R82XX] No valid PLL values for %u Hz!` messages? 
Seeing someone only changed the former text from `[R82XX] PLL not locked!`...

hayguen, 2020-08-01:
the original patch looks to be
https://github.com/mutability/rtl-sdr/commit/f2328161ea9b3d88d5f3b1d76540d9afe7a22c84
from Kyle Keen and Oliver Jowett.

Most of the added/changed code of that patch already is in the development branch here.
But also additional code .. i'm unsure of, how it exactly works.
looking at https://github.com/mutability/rtl-sdr/blob/master/src/tuner_r82xx.c, there are also additional - but different - changes.

someone with more knowledge on the tuner chip is needed!

ValZapod, 2020-08-18:
We implemented it here. https://github.com/librtlsdr/librtlsdr/pull/66

hayguen, 2020-08-18:
cool. thus, i'm closing this.

closed 2020-08-18 by hayguen