librtlsdr/librtlsdr #43: README with high level overview/status for this fork

rct, 2017-02-14:
Since this seems to be the fork of librtlsdr that seems most active (and possibly most up to date), this fork should have a README or other document that gives a high level overview / status of this fork.

Ideally there should be a short summary of the work / other forks that have been integrated into this fork.  For example are keenerd's changes integrated as of a certain date/commit?    

The github commit log is great for the specific details, but is too hard to digest for people trying to figure out which fork of librtlsdr to use.  

Thanks for your work.

E3V3A, 2017-06-30:
Great idea, It seem that Documentation is lacking in pretty much every github repo (including my own).

hayguen, 2020-08-20:
see https://github.com/librtlsdr/librtlsdr/blob/development/README_improvements.md