librtlsdr/librtlsdr #50: rtl_test show actual throughput for determing max usable sample rate

rct, 2017-03-21:
Enhancement request: rtl_test should have a mode where the actual sustained USB throughput is shown to help determine the maximum usable sample rate.  Currently just the number of dropped samples are shown.  Hackrf and airspy have utilities that indicate the transfer rate, which are easily referred to in troubleshooting steps.

E3V3A, 2017-06-30:
I can confirm that this function works poorly for RTL SDRs. No matter what the sample rate set, you get dropped samples, making the test useless.

rct, 2017-06-30:
From my experience, on ports/systems without problems I can run it for reasonable periods with rare/no drops, depending on system/bus activity.  But I think giving betting metrics like throughput rather than sitting there waiting for something to happen would be more meaningful.

hayguen, 2017-06-30:
i've tested multiple dongles with different tuners.
all work without drops up to and including 2400 kSps.
above 2400 kSps you mostly get drops. exceptions are sometimes with very few specific samplerates.

E3V3A, 2017-06-30:
Ok let's be honest here. I'm running this on Kali in a VM on a Windows host. 

**So what causes samples to be dropped? Is it a USB rate/limit or something else?**

So if I am the only one getting this problem, I guess it must be due to latency in the VM/host USB IF. Everything else seem to work great...except that I always get the "PLL is not locked" warning.

hayguen, 2017-07-01:
Please try rtl_test on native host, first!

closed 2020-07-27 by hayguen