librtlsdr/librtlsdr #51: No librtlsdr.dll binary for 64-bit Windows.

chibill, 2017-04-04:
There is no compiled binary (for dll) for 64-bit windows.

This is required by anyone trying to use librtlsdr from python or java when they have a 64 bit version of them installed. 

As librtlsdr is already being compiled for Linux as 64-bit I don't see how compile for windows would be that much of a problem.

chibill, 2017-04-07:
No response?

racerxdl, 2017-04-09:
Sorry, I had been busy these weeks. I will try to make a 64 win build tomorrow. We just didn't done anything because there was no reason todo (most Windows softwares were 32 bit)

chibill, 2017-04-09:
Thanks.

Most windows software is 32 bit because people have no reason to go to 64 bit at the time. But considering you can't buy a 32 bit computer from a name brand anymore I feel like that is going to start changing,

racerxdl, 2017-04-09:
@chibill makes all sense. I would prefer everything x64 like (actually on linux that how it works). I will check if I can build it today and put it here.

racerxdl, 2017-04-09:
@chibill just updated release 0.7.0 to add static librtlsdr for 32 it and 64 bit. Let me know if it works for you: https://github.com/librtlsdr/librtlsdr/releases/tag/v0.7.0

chibill, 2017-04-09:
Thanks. I will test it out as soon as I can.

chibill, 2017-04-09:
Thanks Works perfectly! :P

closed 2017-04-09 by chibill