librtlsdr/librtlsdr #52: Multiple rtlsdr_close issue

jontio, 2017-04-04:
Calling rtlsdr_close stops the data stream callback to function after rtlsdr_open is called again.

This means running str_fm -f 88000000 the first time works but running it a second time no data is returned. The only solution I have found so far is to disconnect the USB dongle then reconnect it again.

My system is a virtual Debian OS running on Windows 7.
Linux debian 3.16.0-4-amd64 #1 SMP Debian 3.16.39-1+deb8u2 (2017-03-07) x86_64 GNU/Linux
VMware® Workstation 12 Player. 12.1.1 build-3770994
Rafael Micro R820T tuner
Realtek, RTL2838UHIDIR 
libusb-1.0 (2:1.0.19-1)

So the following simplified code snippet start/stop can only be called once, run them a second time and "test" wont be displayed again but everything else works as per usual...

```
void rtlsdr_callback(unsigned char *buf, uint32_t len, void *ctx)
{
    qDebug()<<"test";

}

void start()
{
    rtlsdr_open(&rtldev,0);
    rtlsdr_set_sample_rate(rtldev,1200000);
    rtlsdr_set_center_freq(rtldev,88100000);
    rtlsdr_reset_buffer(rtldev);
    future_rtlsdr_read_async_result = QtConcurrent::run(rtlsdr_read_async, rtldev,(rtlsdr_read_async_cb_t)rtlsdr_callback, NULL,0,0);

}

void stop()
{
    rtlsdr_cancel_async(rtldev);
    future_rtlsdr_read_async_result.waitForFinished();
    rtlsdr_close(rtldev);//if this is included it will only run once!!!
    rtldev=NULL;
}
```

If rtlsdr_close never gets called, then when I reopen the rtl device the callback still works.

I've tried a few forks of librtlsdr as well as trying the current master branch of libusb/libusb on githuib and I had the same issue. I'm not sure if it's a problem with libusb, debian, VMware, Windows 7, or librtlsdr. So, a lot of possibilities. Anyway, thought I'd let you know of the strange bug.

jontio, 2017-04-05:
I have been doing a little narrowing down and the problem happens if the the following is written.

`rtlsdr_write_reg(dev, SYSB, DEMOD_CTL, 0x20, 1);`

The code says this is for "poweroff demodulator and ADCs". It seems that once this is written to the Rtl device the ADC never starts again.

This powering down the ADC happens in the `rtlsdr_deinit_baseband(dev)` function that is in turn called when the `rtlsdr_close(dev)` function is called.

hayguen, 2020-08-01:
Hi Jontio,
a modified version of your test program runs just fine:

```
#include <stdio.h>
#include "rtl-sdr.h"

static int rcvCounter = 0;
static rtlsdr_dev_t *rtldev = NULL;

static void rtlsdr_callback(unsigned char *buf, uint32_t len, void *ctx)
{
    ++rcvCounter;
    fputc('#', stderr);
    if (rcvCounter == 5)
    {
        fprintf(stderr, "\ncalling rtlsdr_cancel_async() from within rtlsdr_callback()\n");
        rtlsdr_cancel_async(rtldev);
    }
}

int main(int argc, char **argv)
{
    for ( int k = 0; k < 5; ++k )
    {
        fprintf(stderr, "\nstart no %d ..\n", k);
        rcvCounter = 0;
        rtlsdr_open(&rtldev,0);
        rtlsdr_set_sample_rate(rtldev,1200000);
        rtlsdr_set_center_freq(rtldev,88100000);
        rtlsdr_reset_buffer(rtldev);
        rtlsdr_read_async(rtldev, rtlsdr_callback, NULL, 0, 0 );
        fprintf(stderr, "\n  stopped after %d callbacks.\n", rcvCounter);
        fprintf(stderr, "  closing() ..\n");
        rtlsdr_close(rtldev);
        fprintf(stderr, "  closed.\n");
    }
    return 0;
}
```
i have removed the Qt stuff.
when i have a look at
https://doc.qt.io/qt-5/qtconcurrentrun.html
and
https://doc.qt.io/qt-5/qfuture.html#waitForFinished

probably your understanding of the API call of rtlsdr_read_async() is wrong, as the documentation writes

> Read samples from the device asynchronously. This function will block until it is being canceled using rtlsdr_cancel_async()

meaning, that you need to call rtlsdr_cancel_async() from a concurrent thread or within the callback, cause the callback is called over and over again ..

jontio, 2020-08-02:
Hi hayguen,

My PC setup has changed since I opened this issue.

I cloned built and installed this repo then tested your test code above and it worked as expected opening reading and closing the rtl device.

I then tried the a qt project using the code that I posted above and it too worked (it's just a form with 2 push buttons on it)

```C++
#include "mainwindow.h"
#include "ui_mainwindow.h"
#ifdef __cplusplus
extern "C" {
#endif
#include <rtl-sdr.h>
#ifdef __cplusplus
}
#endif
#include <QDebug>
#include <QtConcurrent/QtConcurrent>
void rtlsdr_callback(unsigned char *buf, uint32_t len, void *ctx)
{
    qDebug()<<"test";
}
rtlsdr_dev_t *rtldev;
QFuture<void> future_rtlsdr_read_async_result;
void start()
{
    rtlsdr_open(&rtldev,0);
    rtlsdr_set_sample_rate(rtldev,1200000);
    rtlsdr_set_center_freq(rtldev,88100000);
    rtlsdr_reset_buffer(rtldev);
    future_rtlsdr_read_async_result = QtConcurrent::run(rtlsdr_read_async, rtldev,(rtlsdr_read_async_cb_t)rtlsdr_callback, NULL,0,0);
}
void stop()
{
    rtlsdr_cancel_async(rtldev);
    future_rtlsdr_read_async_result.waitForFinished();
    rtlsdr_close(rtldev);//if this is included it will only run once!!!
    rtldev=NULL;
}
MainWindow::MainWindow(QWidget *parent)
    : QMainWindow(parent)
    , ui(new Ui::MainWindow)
{
    ui->setupUi(this);
}
MainWindow::~MainWindow()
{
    delete ui;
}
void MainWindow::on_pushButton_clicked()
{
    start();
}
void MainWindow::on_pushButton_2_clicked()
{
    stop();
}
```

`QtConcurrent::run` will call the the function as another thread so it doesn't block and you can stop it using the main thread.

So it looks like the issue has only happened on the old computer using the usual setup I mentioned before. I wounder if the problem lies with VMWare. I'll dig out the old computer and see if I can get it working. The only other odd thing with that computer is that it has none of the AVX instructions. My current setup that I tested both your code and mine is...

Realtek, RTL2838UHIDIR
Rafael Micro R820T tuner
libusb 2:1.0.23-2build1 amd64
Linux 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Cheers,
Jonti

jontio, 2020-08-05:
I dug out the old Windows computer (Windows 7 Ultimate, 64-bit 6.1.7601, Service Pack 1) and setup the following...

VMware® Workstation 15 Player 15.5.6 build-16341506
Linux debian 4.19.0-10-amd64 #1 SMP Debian 4.19.132-1 (2020-07-24)
libusb-1.0 (2:1.0.22-2)

This time running your code it froze at the second try...

```BASH

start no 0 ..
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
#####
calling rtlsdr_cancel_async() from within rtlsdr_callback()

  stopped after 5 callbacks.
  closing() ..
  closed.

start no 1 ..
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
#

```

hayguen, 2020-08-05:
i just made a similar setup with more up-to-date components:
* Windows 10 as host
* Oracle VirtualBox 6.1
* Raspberry OS, a derivative of debian/GNU Linux 10 (buster)

this is the output, i get:
```
pi@ScanVM:~/librtlsdr/test $ ./test_open_after_close

start no 0 ..
Found Rafael Micro R820T tuner
[R82XX] PLL not locked for 2043000 Hz!
Allocating 15 zero-copy buffers
#####
calling rtlsdr_cancel_async() from within rtlsdr_callback()

  stopped after 5 callbacks.
  closing() ..
  closed.

start no 1 ..
Found Rafael Micro R820T tuner
[R82XX] PLL not locked for 2043000 Hz!
Allocating 15 zero-copy buffers
#####
calling rtlsdr_cancel_async() from within rtlsdr_callback()

  stopped after 5 callbacks.
  closing() ..
  closed.

start no 2 ..
Found Rafael Micro R820T tuner
[R82XX] PLL not locked for 2043000 Hz!
Allocating 15 zero-copy buffers
#####
calling rtlsdr_cancel_async() from within rtlsdr_callback()

  stopped after 5 callbacks.
  closing() ..
  closed.

start no 3 ..
Found Rafael Micro R820T tuner
[R82XX] PLL not locked for 2043000 Hz!
Allocating 15 zero-copy buffers
#####
calling rtlsdr_cancel_async() from within rtlsdr_callback()

  stopped after 5 callbacks.
  closing() ..
  closed.

start no 4 ..
Found Rafael Micro R820T tuner
[R82XX] PLL not locked for 2043000 Hz!
Allocating 15 zero-copy buffers
#####
calling rtlsdr_cancel_async() from within rtlsdr_callback()

  stopped after 5 callbacks.
  closing() ..
  closed.
```

probably, i should test exactly your program?
would you send me your CMakeLists.txt and which packages i need to install?

jontio, 2020-08-06:
I've just tried virtualbox rather than vmware with the same physical hardware as before and that didn't cause the issue. I used the same iso and the same setting up of the guest system as I did with vmware the other night.

In both cases my setup of the guest is almost identical; it goes like this...

debian-live-10.5.0-amd64-xfce.iso.torrent

choose debian installer

lots of pressing enter bar for user name of jontio and password things

```
su -
usermod -aG sudo jontio
```
log out then in again

```
sudo apt update
sudo apt upgrade
sudo apt install git cmake libusb-1.0-0-dev
mkdir git
cd git
git clone https://github.com/librtlsdr/librtlsdr.git
cd librtlsdr
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
sudo cp ../rtl-sdr.rules /etc/udev/rules.d/
echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist.conf
sudo reboot
```

```
nano test.c
```
copy and pates your code. exit nano and save.
```
gcc test.c -lrtlsdr
./a.out
```
For tonight's test with virtualbox (doesn't cause the issue) I installed ...

`VirtualBox-6.1.12-139181-Win.exe`

`Oracle_VM_VirtualBox_Extension_Pack-6.1.12.vbox-extpack`

and the VM settings of...

```
new linux debian 64bit os 1601MB ram 32GB VDI drive dynamic allocation, USB 2 (OHCI + EHCI), chipset PIIX3, processors 2, 128MB video ram VMSVGA
```

The only other thing I installed on the vmware version that wasn't on the virtualbox version were the guest addons `sudo apt install open-vm-tools`.

So it's looking like so far it only happens in vmware and on the computer with the G3220 CPU. It will be interesting to see if you get the issue when using vmware. I've got a newer computer I can install windows 10 on (AMD Ryzen 3400G) and then test both vmware and virtualbox on that.

jontio, 2020-08-07:
Hi hayguen,

I just tried it with VMware (Workstation 15 Player 15.5.6 build-16341506) on the new computer Ryzen 3400G, 16GB, ASUS prime B450M-A windows 10 host (version 2004 build 19041.388), same guest os install as before and your test code froze on the second opening of the rtl device. So it looks like what's important is running VMware. I have one other rtl device somewhere, I'll try that tomorrow.

Cheers,
Jonti

hayguen, 2020-08-07:
i'll try WMware also .. but need some time until doing that ..
if one application freezes .. does it help to disconnect/reconnect the RTLSDR device "virtually" from the VMware session, to allow running it once again?

jontio, 2020-08-07:
Yup disconnect then reconnect the RTLSDR device "virtually" in VMware allows it to run again.

hayguen, 2020-08-18:
installed Ubuntu 20.04 LTS with latest updates on a Windows 10 host (also with latest updates)
in a VMware Workstation 15 Player version 15.5.6 build-16341506 (same as yours).

then ran the test program and everything worked fine .. with 2 differerent RTL dongles - both with R820T2 tuner.

jontio, 2020-08-18:
Oh that is strange. I found the other RTL dongle and have now tested them both. They both enumerate as `0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T`, librtlsdr says `Found Rafael Micro R820T tuner` and both of them fail with your test code with vmware on both windows computers.

I tried Ubuntu 20.04.1 LTS just in case but that behaved just the same debian xfce 10.5.0. and that failed.

hayguen, 2020-08-18:
yes, really strange.
can someone else test this issue?

ValZapod, 2020-08-24:
From what I read it is just a Vmware issue. Vmware is very buggy, for example, for me it causes deadlocks once in a while and also it loses ip prefixes (only ipv4, ipv6 is good) on internal virtual eth0.

hayguen, 2020-09-03:
fear, we don't get further insights .. agree to close this issue?

jontio, 2020-09-03:
Yup sounds good.

closed 2020-09-03 by jontio