librtlsdr/librtlsdr #55: Upstream patches from this fork

kerel-fs, 2017-06-14:
Hi,
since the upstream is active, I would like to bring relevant patches from this fork upstream.
The bias-tee support was merged a few days ago, but a huge list of commits is with potential patches is still available (see below).

A short list of available patches:
- [x] Add bias-tee support
- [ ] Add IrDA, IR support and rtl_ir
- [ ] Add rtlsdr_rpc

Detailed list:
- [ ] 4a0e25b added option -L to print current level values added option -c to configure de-emphasis time constant for wbfm added option -v to print verbose (debug) output added alias modulation names nbfm, nfm, wfm

- [ ] 5e063d8 "rdc" filter in rtl_fm now at capture rate, before mixing or filtering:   DC is now totally removed, when not using option  "-F 9"   due to bad anti-alias rejection the RTL dongle's DC folded into the decimated band   therefore swapped conversion to 16 bit and up-mixing by fs/4 (rotate_90)

- [ ] f5c7ad9 added dc filter on raw data (after decimation) with option "-E rdc"   added option "-q" for setting averaging speed   added option "-E adc" in addition to "-E dc"

- [ ] aebbb9a Added "Hz" in bandwidth set message.
- [ ] 9a2c474 Device Bandwidth to Convenience and rtl_fm
- [ ] ba43a14 Updates on rtl_tcp

- [ ] af20337 added rtlsdr_set_and_get_tuner_bandwidth() to RTLSDR_API
- [ ] 32b051b Added default bandwidth to list

- [ ] aabe859 rtl_fm: made -v flag without argument rtl_tcp: added verbose option -v rtl_tcp: reduced mutex timeout from 5 to 1 sec.   makes rtl_tcp re-available faster fixed formatting/indentation with tabs
- [ ] 5723807 merged option to activate digital agc of rtl2832 with '-E rtlagc'
- [ ] 240abf6 use offset tuning when using bandwidth option

- [ ] 20bc4dc added CMakeLists.txt for Win32 and a README
- [ ] 35639ad minor correction in usage

- [ ] e6bdc91 added option '-l' for rtl_tcp to set block length in units of 512 samples

- [ ] 3cda64c Bug Fix for some Broken MinGW PThreads sources

- [ ] 94a55d9 Fix clang warning about shifting a negative signed value

- [ ] c4e955c Only read if IR_RX_BC returns 0x83, to match Linux DVB driver
- [ ] 3ae2072 Do not read when IR_RX_IF returns 0x82/1, fixes -7 libusb timeout on RPi
- [ ] c45833a Fix compile warnings with GCC 4.9.2
- [ ] 0cd8007 rtl_tcp: add -I to enable infrared listener port (rtl_ir)
- [ ] b446cd1 Read data when IR_RX_IF returns 0x82, 0x81 not only 0x83
- [ ] e493671 rtl_ir: flush output after writing (so can use with |xxd -b)
- [ ] 61d845e rtlsdr_ir_query() returns packed buffer, rtl_ir prints
- [ ] ed6c2de rtl_ir: query IR in a loop
- [ ] 05649bf Add rtlsdr_read_regs to read IR data into buffer
- [ ] d7a7a08 Fix IR block claculation for read/write registers non-array
- [ ] 5824c16 Rename rtl28xxu_wr_reg_mask -> rtlsdr_write_reg_mask
- [ ] 45abfbe Fix rtl28xxu_wr_reg_mask return code, now <0 is error
- [ ] 70a34fd Begin integrating IR from dvb_usb_rtl28xxu into librtlsdr
- [ ] 4ca84ae Add infrared block USB control message indexes
- [ ] 4b4942f rtl_ir: merge some init code from Linux dvb-usb-v2 driver
- [ ] 024f3c5 rtl_ir: initialize dongle
- [ ] aa206d0 Add rtl_ir tool stub
- [ ] 0693626 Add IrDA and IR registers from Linux dvb_usb_rtl28xxu driver

- [ ] 8a68ce1 add uninstall note to README.rtlsdr_rpc
- [ ] 3be49cb update list of known working software, gqrx
- [ ] b80d5c9 update list of known working software, cubicsdr
- [ ] 8f04344 add list of known working software
- [ ] 078a487 handle conccurrent requests
- [ ] 4fb1ead note abount using latest libusb
- [ ] b321103 send_flush_msgs
- [ ] 4008c0b default gain_count
- [ ] 0d72bb4 fix reply offset
- [ ] 91e085b fix get_tuner_gains
- [ ] 3796c23 fix get_tuner_gains
- [ ] 055403b add README.rtlsdr_rpc
- [ ] 2cb35d0 default server address to wildcard
- [ ] 5cfd82c fixme: usleep
- [ ] ad0cc0a remove some PRINTF
- [ ] 431631c rtlsdr_rpc_get_offset_tuning
- [ ] dab5cb6 rtlsdr_rpc_set_offset_tuning
- [ ] d40cd42 rtlsdr_rpc_get_direct_sampling
- [ ] f14c3b5 rtlsdr_rpc_set_direct_sampling
- [ ] f5f543e rtlsdr_rpc_set_agc_mode
- [ ] a08dd08 missing tuner routines
- [ ] c39f7f0 rtlsdr_rpc_set_testmode
- [ ] 245819d fflush stderr. uint8_t fmt.
- [ ] 2f0ecfe ok if device already opened
- [ ] 5cffcda implement rtlsdr_rpc_get_usb_strings
- [ ] a5e0e0e add rtlsdr_rpc support

- [ ] 110c4a5 rtl_rpcd: accept command-line arguments

- [ ] 2316a13 rtl_rpcd: on IR thread, use select() to check if can send instead of blocking forever
- [ ] 04992ac rtl_rpcd: add IR listening option

- [ ] 7228e75 Sorry, wrong notation at CMAKE
- [ ] f0c0e8a Disabling RPC in Windows builds. It uses POSIX calls and we should port it in the future.

- [ ] e35935f Fix misc. compilation issues, update gitignore
- [ ] 35bf7ad minor fix

- [ ] 23ce5dd fix rtl_tcp's default bufferlength for smooth streaming

- [ ] a82583 scanning via csv command file with specific gain and command-execution triggers
- [ ] 514924 allow units for samplerate with rtl_test option '-s'
- [ ] 286059 silence compiler warning, complaining C90 declaration after statement
- [ ] 3c4028 direct sampling mode switching depending on center_frequency

- [ ] 4f1bdf8 added options adcmax and adcrms in command file and blocksize

- [ ] 30a3216 Moved include rtl_tcp.h from rtl-sdr.h to rtl_tcp.c (where it should be)
- [ ] 90afc43 Added option '-w' to rtl_sdr that allows setting the tuner bandwidth.

- [ ] 0c27010 Fixed missing stddef.h in rtl-sdr.h

- [ ] c20ba84 Fix check for NEED_PTHREADS_WORKARROUND
- [ ] 51ae5b5 rtl_fm: snprintf for older MSVC
- [ ] 9b03f75 rtl_ir: round() provided by newer MSVC
- [ ] acc3126 rtl_tcp: usleep definition for windows

kerel-fs, 2017-06-14:
This issue is related to #43: When we know what differs between this fork&the upstream, we can try upstreaming this differences if preferable.

hayguen, 2020-09-03:
if someone has interest to do so .. please create issue(s) at upstream!

closed 2020-09-03 by hayguen