librtlsdr/librtlsdr #56: Compilation fails on Cygwin x86 (32 bit)

K2DLS, 2017-08-04:
[ 33%] Linking C executable rtl_power.exe
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x1bb): undefined reference to `rtlsdr_set_tuner_gain_mode'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x1d1): undefined reference to `rtlsdr_get_tuner_gains'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x1f2): undefined reference to `rtlsdr_get_tuner_gains'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x3ae): undefined reference to `rtlsdr_set_and_get_tuner_bandwidth'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x495): undefined reference to `rtlsdr_set_direct_sampling'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x594): undefined reference to `rtlsdr_set_offset_tuning'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x684): undefined reference to `rtlsdr_set_tuner_gain_mode'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x705): undefined reference to `rtlsdr_set_tuner_gain_mode'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x719): undefined reference to `rtlsdr_set_tuner_gain'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x7e1): undefined reference to `rtlsdr_set_freq_correction'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x84c): undefined reference to `rtlsdr_reset_buffer'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x89b): undefined reference to `rtlsdr_get_device_count'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x900): undefined reference to `rtlsdr_get_device_usb_strings'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x98f): undefined reference to `rtlsdr_get_device_usb_strings'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0x9cb): undefined reference to `rtlsdr_get_device_usb_strings'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0xa2d): undefined reference to `rtlsdr_get_device_usb_strings'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0xaa6): undefined reference to `rtlsdr_get_device_name'
libconvenience_static.a(convenience.c.o):convenience.c:(.text+0xada): undefined reference to `rtlsdr_get_device_name'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/rtl_power.dir/build.make:98: src/rtl_power.exe] Error 1
make[1]: *** [CMakeFiles/Makefile2:159: src/CMakeFiles/rtl_power.dir/all] Error 2
make: *** [Makefile:128: all] Error 2

hayguen, 2020-08-01:
suppose nobody is still working on cygwin .. can we close this?

K2IE, 2020-08-01:
Your choice.  Since no one else has commented in 3 years I don't think it is an issue for many!

closed 2020-08-01 by hayguen