librtlsdr/librtlsdr #57: [R82XX] PLL not locked!

attie, 2017-10-22:
I've just tracked down an issue related to the PLL failing to lock, and thus decoding corrupt data.
It was failing on my new system (I build librtlsdr from source), and working fine on my old system (Ubuntu 14.04, using the `librtlsdr0` package, Version: 0.5.2.7.3ab6-1).

A bisect has shown up d892279085491f9c610388da89bc701ef6ac632c as the first bad commit.

I'm somewhat suprised by this, given the commit's age (~2 years 7 months), but reverting to its parent (d447a2e) has solved the problem for me.

I will look into this further when I can, but for now will be using d447a2e.

I'm running Ubuntu 16.04.3 on an i7-7700K.

attie, 2017-10-22:
Package info from functional Ubuntu 14.04 on i7 2600

```
$ apt-cache show librtlsdr0
Package: librtlsdr0
Priority: extra
Section: universe/libs
Installed-Size: 114
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Adam Cécile (Le_Vert) <gandalf@le-vert.net>
Architecture: amd64
Source: rtl-sdr
Version: 0.5.2.7.3ab6-1
Depends: libc6 (>= 2.14), libusb-1.0-0 (>= 2:1.0.8)
Pre-Depends: multiarch-support
Filename: pool/universe/r/rtl-sdr/librtlsdr0_0.5.2.7.3ab6-1_amd64.deb
Size: 26488
MD5sum: 928419846cc69885ff51fbda362a36b5
SHA1: 9605b19c22ed1d28d6de8d6a1f39281df2039253
SHA256: 01574a47a9867a37c2a8ff11fff65ba9920ab0e6ae76f5da754144a86d130c94
Description-en: Software defined radio receiver for Realtek RTL2832U (development files)
 rtl-sdr is a software defined radio (SDR) receiver software for certain
 low-cost DVB-T/DAB(+) USB dongles based on the Realtek RTL2832U chip.
 .
 This package contains development files.
Description-md5: 83ca0f034542873ebb35fc804478ed50
Multi-Arch: same
Homepage: http://sdr.osmocom.org/trac/wiki/rtl-sdr
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Origin: Ubuntu
```

attie, 2017-10-22:
Problems on Ubuntu 16.04 on i7-7700K (built from c7d970a)
```
$ ./rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
```

Less(?) problems on Ubuntu 16.04 on i7-7700K (built from d447a2e)
```
$ ./rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
```

attie, 2017-10-25:
Further investigation shows that the `[R82XX] PLL not locked!` does in fact not appear to be a problem.
Actually, moving the antenna a couple of meters allowed me to receive a signal from a neighbour as well as mine...

I have had my application running on the i7-7700k with c7d970a for the past few days, and the data points coming from two(!) devices are stable.

I don't think closing the issue is appropriate yet, as running on the i7-2600 with c7d970a also shows the `[R82XX] PLL not locked!` message.

* Can this message be safely ignored?
* How come it's started showing up since d892279?
* What can we do to fix the issue?

wgaylord, 2018-01-06:
I found the cause of this to actually have to do with the calling order of some functions. (A Python library using librtlsdr found this.)
https://github.com/roger-/pyrtlsdr/issues/72 Shows that the calling order generates this. Not sure if its easily fixable here.

wgaylord, 2018-02-06:
I have a reason why it began happening. I believe it is because the bandwidth is being set before the if freq. Which causes the chip to think you are doing stuff before the PLL has stetted and locked to the freq.

Shouldn't be an issue but I can make a PR to fix this. (Simple changing the order of calls in the r820t_set_bw function.)

attie, 2018-02-06:
Interesting, thanks for the update.

I have been using things now since my last post with the "_error_" and without issues.

hayguen, 2020-08-01:
i am ignoring this warning ever since ..
is there a pull request for latest development branch?
can we simply close this issue?

ValZapod, 2020-08-17:
Yep I also got such warnings.))

ValZapod, 2020-08-17:
You should apply this patch https://github.com/librtlsdr/librtlsdr/commit/c28886f89b24e4cc73541710b2d6e6c013ed90c5 to check what are the freq. as BTW recommended here and is done in tuner_e4k.c and shown here https://inst.eecs.berkeley.edu/~ee123/fa13/rtl_sdr_install.html, I am also almost sure that it is just a low freq. that is not supported, not because of r820t_set_bw function. LOL.

hayguen, 2020-08-18:
ok, that patch https://github.com/librtlsdr/librtlsdr/commit/c28886f89b24e4cc73541710b2d6e6c013ed90c5 is already in the development branch

ValZapod, 2020-08-18:
> already in the development branch

But it is not in bisected tree before the breaking  commit d892279085491f9c610388da89bc701ef6ac632c (that is why I am "almost sure").

ValZapod, 2020-08-18:
Both kal -s GSM900 and rtl_test -p do not print that warning before https://github.com/librtlsdr/librtlsdr/commit/d892279085491f9c610388da89bc701ef6ac632c (as obviously both rtl_test and kal cannot set bandwidth before that, for 828D or 820T). 

NOW WHEN YOU checkout https://github.com/librtlsdr/librtlsdr/commit/d892279085491f9c610388da89bc701ef6ac632c and apply https://github.com/librtlsdr/librtlsdr/commit/c28886f89b24e4cc73541710b2d6e6c013ed90c5 on top it prints 

```
root@kali:~/# kal -s GSM900 -e 29 //-e is error in ppm
Found 1 device(s):
  0:  Astrometa DVB-T/DVB-T2

Using device 0: Astrometa DVB-T/DVB-T2
Found Rafael Micro R828D tuner
Exact sample rate is: 270833.002142 Hz
[R82XX] PLL not locked for 2125000 Hz!
kal: Scanning for GSM-900 base stations.
GSM-900:
	chan: 25 (940.0MHz + 1.730kHz)	power: 179537.60
	chan: 43 (943.6MHz - 1.831kHz)	power: 177609.95
```

So it is not a bug in the code. 2.125 000 MHz is not supported. LOL! And then **I just add else with "PLL locked"** 
```
diff --git a/src/tuner_r82xx.c b/src/tuner_r82xx.c
index 0f84a0d..36ed3c9 100644
--- a/src/tuner_r82xx.c
+++ b/src/tuner_r82xx.c
@@ -550,7 +550,7 @@ static int r82xx_set_pll(struct r82xx_priv *priv, uint32_t freq)
                fprintf(stderr, "[R82XX] PLL not locked for %u Hz!\n", freq);
                priv->has_lock = 0;
                return 0;
-       }
+       } else fprintf(stderr, "[R82XX] PLL locked for %u Hz!\n", freq); //dumb, it is not stderr, but whatever
 
        priv->has_lock = 1;

```

and got

```
root@kali:/opt/SourceCode/librtlsdrnew# kal -s GSM900
Found 1 device(s):
  0:  Astrometa DVB-T/DVB-T2

Using device 0: Astrometa DVB-T/DVB-T2
Found Rafael Micro R828D tuner
[R82XX] PLL locked for 56000000 Hz! //supported
Exact sample rate is: 270833.002142 Hz
[R82XX] PLL not locked for 2125000 Hz! //not supported
kal: Scanning for GSM-900 base stations.
[R82XX] PLL locked for 937325000 Hz!
[R82XX] PLL locked for 937525000 Hz!
[R82XX] PLL locked for 937725000 Hz!
[R82XX] PLL locked for 937925000 Hz!
[R82XX] PLL locked for 938125000 Hz!
[R82XX] PLL locked for 938325000 Hz!
[R82XX] PLL locked for 938525000 Hz!
[R82XX] PLL locked for 938725000 Hz!
[R82XX] PLL locked for 938925000 Hz!
[R82XX] PLL locked for 939125000 Hz!
[R82XX] PLL locked for 939325000 Hz!
```

hayguen, 2020-08-18:
hope, latest patch on the development branch should fix this:
* looks, calling r82xx_set_pll() at initialization before having called r82xx_set_freq() produced the erroneous warning
* now just remembering last rf_freq given at r82xx_set_freq() to suppress the warning

tested with kal and rtl_fm: both look good

ValZapod, 2020-08-18:
But priv->has_lock = 0;
still has no lock, it cannot lock on such a low freq?? See my previous message.

Also did not it introuduce dead code??? With if ( priv->rf_freq )

Also you have tabs issues the first comment))

hayguen, 2020-08-19:
2nd try:
* at initialization from r82xx_set_tv_standard() the frequency wasn't set from outside. fixed this.
* suppression removed
* yes, for low frequencies (below 24 MHz for R820T), the PLL can't lock
* setting bandwidth - before having set frequency - provoked above error. now keeping previous frequency in r82xx_set_pll() when called to set 0.

ValZapod, 2020-08-19:
>  now keeping previous frequency in r82xx_set_pll() when called to set 0.

So it is this after all:
> Simple changing the order of calls in the r820t_set_bw function

hayguen, 2020-08-19:
No, changing the call order is not always the solution.
depending on the rtl tool bandwidth is set before setting the frequency ..

ValZapod, 2020-08-19:
Okay, last question, why with kal it is:
[R82XX] PLL not locked at Tuner LO **1925000** Hz!

and with rtl_test -p it is 
[R82XX] PLL not locked at Tuner LO **1530000** Hz!

(is it because of **uninitialized variable** that is sending garbage, really)? Otherwise looks good to me, please pull https://github.com/steve-m/librtlsdr/commit/be1d1206bfb6e6c41f7d91b20b77e20f929fa6a7 as it is a memory leak.

hayguen, 2020-08-19:
> (is it because of uninitialized variable that is sending garbage, really)?

Don't think so: internal tuner frequency depends on bandwidth and automatic bandwidth depends on Samplerate. The different applications probably use different Samplerates ..
Will check that again .. tomorrow.

Memory leak: thanks for reminder.

hayguen, 2020-08-19:
kal does use 270833 Hz samplerate.

when calling rtl_test with same samplerate:
```
$ ./rtl_test -s 270833 -p
..
[R82XX] PLL locked at Tuner LO 56000000 Hz for RF 56000000 Hz!
..
r82xx_set_freq(freq = 56000000) @ LSB--> intfreq 1925000, ifcenter 0 --> f 57925000
[R82XX] PLL locked at Tuner LO 57925000 Hz for RF 56000000 Hz!
```
you can see intfreq of 1925 kHz - same as your not locked Tuner LO with kal.
thus, we don't have an uninitialized variable.

hayguen, 2020-08-19:
> please pull steve-m@be1d120 as it is a memory leak.

cherry-picked that patch

ValZapod, 2020-08-19:
Ok, if you are sure PLL lock in place where PLL lock was absent will not cause any problems, it is LGTM.

closed 2020-09-03 by hayguen