librtlsdr/librtlsdr #58: How to open 2 devices (Question)

natiya, 2017-11-16:
I'm trying to open 2 USB dongles but I can't get it to work. When I do 

```
    a=librtlsdr.rtlsdr_get_device_name(1)
    print(a)
    b=librtlsdr.rtlsdr_get_device_name(3)
    print(b)
    librtlsdr.rtlsdr_open(a,1)
```

I get: 

> b'Generic RTL2832U OEM'
> b'Generic RTL2832U OEM'
> Traceback (most recent call last):
>   File "C:\Users\...\example.py", line 23, in <module>
>     main()
> File "C:\Users\...\pyrtlsdr-master\example.py", line 21, in main
>     librtlsdr.rtlsdr_open(a,1)
> ctypes.ArgumentError: argument 1: <class 'TypeError'>: expected LP_c_void_p instance instead of bytes

I don't know why I get the same name twice, but this is not the error I'm getting. Any suggestion?

racerxdl, 2017-11-17:
As I remember the rtlsdr_open calls doesn't use the device name parameter.

natiya, 2017-11-17:
I found a way to do it:
```
from rtlsdr import RtlSdr
    sdr = RtlSdr(serial_number='x1')
    sdr2 = RtlSdr(serial_number='x2')
```

closed 2017-11-17 by natiya