librtlsdr/librtlsdr #60: Cannot find tuner

js3263854, 2018-02-07:
I've build rtl_entropy with the latest librtlsdr and contacted the author and they stated that the issue could be with librtlsdr. Any ideas?

foo@bar:~$ lsusb
Bus 001 Device 004: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

foo@bar:~$ rtl_entropy
Options parsed, continuing.
Found 1 device(s):
  0:  Generic RTL2832U OEM
Using device 0: Generic RTL2832U OEM
No supported tuner found
Enabled direct sampling mode, input 1
Exact sample rate is: 2675000.143941 Hz
Setting Frequency to 101500000
Your device is capable of gains at...
 : 0.00
Setting gain to 0.00
Doing FIPS init
Reading samples in sync mode...

itdaniher, 2018-02-07:
"tuner" in this case refers specifically to the frontend chip on your receiver, typically an R820T or a E4000 part, not to the USB stick as a whole.

If it's trivial to look at the circuit board inside your USB stick, you might have luck opening it and looking at the chip closest to your RF connection port. 

For what it's worth, for use as an entropy source, I don't think the tuner part itself is required - you should be able to get good noise by setting a high gain on the RTL2832U itself, but this would likely require source modifications to rtl_entropy.

js3263854, 2018-02-07:
It is an R820T. I'm starting to think its DOA.

closed 2020-07-27 by hayguen