librtlsdr/librtlsdr #64: Question about your Hann-Poisson implementation:

Kishaihd, 2018-12-17:
This is more of a question if this was done intentionally in code.
I noticed that in the Hann-Poisson windowing function, you used `1 - cos()`, but mathematically, this window function uses `1 + cos()`. Just wondering why you did it this way, and if there's a benefit to doing `1 - cos()`?

Kishaihd, 2018-12-19:
Interestingly, the Stanford University website is the only place I've found the Hann-Poisson window function to be defined as `1 + cos()`. 
Every other resource I've found including the Matlab wiki shows it as `1 - cos()`. And Matlab seems to be a very trustworthy source. 
So, never mind on my comment, I guess?

hayguen, 2018-12-19:
found it on stanford site and dsprelated with 1 + cos():
https://ccrma.stanford.edu/~jos/sasp/Hann_Poisson_Window.html
https://www.dsprelated.com/freebooks/sasp/Hann_Poisson_Window.html

and on wikipedia with 1 - cos:
https://en.wikipedia.org/wiki/Window_function#Hann%E2%80%93Poisson_window

BUT: check the argument of the cos()! you see it's different .. as the exponent.

That is because both define w(n) but one of the definitions is time shifted, starting at the peak in the center, the other definition .. not. the symmetry of the windows allows this.