librtlsdr/librtlsdr #70: How to support "from rtlsdr import RtlSdr" mentioned on https://pypi.org/project/pyrtlsdr/?

krhohio, 2019-06-18:


krhohio, 2019-06-18:
Helping a friend with this and I have not passed "Python 101" yet.
Should there be a rtlsdr.dll available somewhere?

Here's the pyrtlsdr text:
**Dependencies**
Windows/Linux/OSX
Python 2.7.x/3.3+
librtlsdr
Optional: NumPy (wraps samples in a more convenient form)
matplotlib is also useful for plotting data. The librtlsdr binaries (rtlsdr.dll in Windows and librtlsdr.so in Linux) should be in the pyrtlsdr directory, or a system path. Note that these binaries may have additional dependencies.

hayguen, 2020-07-27:
ouch: this issue is quite old .. and nobody responded. sorry.

concerning "should there be rtlsdr.dll somewhere":
i think, for pyrtlsdr: yes.

(i've not tried/worked with pyrtlsdr)
what exactly is the issue?

i fear you need to talk with the pyrtlsdr guys for help .. or to support this librtlsdr fork, with it's additional functionality (mostly still in the development branch).

closed 2020-08-19 by hayguen