librtlsdr/librtlsdr #76: new release

hayguen, 2020-08-18:
i'd like to make a new release with all the changes/enhancements/features from current development branch.
for such a new release, all issues and pull requests should get processed/closed or at least reviewed, within reasonable time.
**i'd need some help on this**

MartinHerren, 2020-09-27:
Hello, what is the status of his fork compared to the steve-m/librtlsdr ?
It would definitively be good to have a single 'official' and maintained source for rtl_sdr and definitively a new release with proper versioning would be good.

I've only used the rtl_sdr driver in some projects, didn't dig a lot into the internals (just enough to understand how to use the api when it was unclear). But if there is something i could help with i could find some time.

hayguen, 2020-09-27:
for a status, have a look at
1) https://github.com/librtlsdr/librtlsdr/blob/development/README_improvements.md
2) open and close issues https://github.com/librtlsdr/librtlsdr/issues marked with "next major release".

the README_improvements.md meanwhile needs an update:
* this fork supports harmonic reception up to and possibly over 6 GHz. this works in 'lab' .. but real world will need a band- or at least highpass filter (SAW?) in front and also an additional LNA to compensate loss

> It would definitively be good to have a single 'official' and maintained source for rtl_sdr

words are nice .. but when you look at github, there are many 'famous' forks around, which were not merged into osmocom or steve-m - if you regard them as 'official'.
this fork does welcome pull requests .. and many patches were integrated, see https://github.com/librtlsdr/librtlsdr/issues/34 and especially https://github.com/librtlsdr/librtlsdr/issues/17

> But if there is something i could help with i could find some time.

thank you:
* simply have a look at the issues. you see, that most are meanwhile solved and marked with 'verify'. but some issues could need some more testing.
* i tried to document the API a bit better .. you might have a look at that also .. and probably point out unclear things

something missing:
* automatic compilation/build for windows. linux and macos do build automatically utilizing github actions

hayguen, 2020-10-04:
i've updated the docs today.
think, we are sort of feature complete and stable enough to release.
any thoughts on this?

hayguen, 2020-10-07:
declared to release v0.8

closed 2020-10-07 by hayguen