librtlsdr/librtlsdr #77: high samplerates without loss

hayguen, 2020-08-19:
from my tests with several RTLSDR dongles, good samplerates are up to 2400 kSps.
some days ago, i could verify that the samplerates
```  f(k) = 2400 + k * 40 kHz```
were also good .. up to 2880 kHz. that was on Raspberry Pi 4.

today, when running
```
$ rtl_test -s 2440k
..
Sampling at 2440000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 168 bytes
lost at least 68 bytes
lost at least 188 bytes
lost at least 68 bytes
lost at least 68 bytes
lost at least 68 bytes
lost at least 68 bytes
lost at least 188 bytes
lost at least 68 bytes
^CSignal caught, exiting!
```
i get multiple, "lost at least .." messages. just to remind, one single/initial "lost at least .." is OK.
but this time, i ran rtl_test for several samplerates in a Linux (Ubuntu 20.04LTS) VM, hosted by Windows 10.

having a continous stream, without lost samples, is very important for decoding digital modes, with fixed symbolrate.
even the results of rtl_power might get distorted - especially when the losses get too frequent.

can other people try samplerates above 2400 kHz with rtl_test and report the results?

racerxdl, 2020-08-19:
I usually use 2560khz which seens the most stable one for my dongles (more than 2400) (even the cheap chinese ones). Can you try that and see if there is some loss?

hayguen, 2020-08-19:
that looks also good for me - even in the VM:
```
$ time rtl_test -s 2560k
..
Sampling at 2560000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 648 bytes
lost at least 340 bytes
lost at least 207 bytes
^CSignal caught, exiting!

User cancel, exiting...
Samples per million lost (minimum): 3

real	1m4.722s
user	0m0.084s
sys	0m0.209s
```
the 3 lost messages are all directly at start

ValZapod, 2020-08-21:
This haens because of zero copy stuff, no? See also https://github.com/steve-m/librtlsdr/commit/f68bb2fa772ad94f58c59babd78353667570630b

ValZapod, 2020-08-24:
> rtl_test -s 2560k -p

In Vmware: interesting, this works correctly with only one "lost at least 152 bytes in buffer 0", rtl_test -s 2440k -p indeed cause 3 messages like that... Hmmm.

Will check when I will reboot my virtual machine into real one...

merbanan, 2020-08-26:
I have an older motherboard that can sample at 3.2M with very few dropped packets (ie a digital demod can resync). It depends on the usb controller chip, the reason for the first dropped packet can be because of some internal buffering in the bridge chip.

ValZapod, 2020-08-26:
> motherboard

I.e. Intel/Amd motherboard or motheboard of Rtl-sdr?

> of some internal buffering in the bridge chip.

Or drivers. Or Intel ME.

merbanan, 2020-08-26:
It was/is an Intel motherboard.