librtlsdr/librtlsdr #83: PLL frequency and compensation

hayguen, 2020-08-24:
it would be nice to have a good/automatic test procedure to create a plot as in for steve-m#10
the patch for that was merged with PR: #66

there must be a test/detection program for detecting the frequency of this GSM 954.0 MHz FCCH burst!? does kal use that?

in addition, the PLL error might be compensated by setting/fixing the IF frequency, the RTL2832 is programmed to.

ValZapod, 2020-08-24:
@mutability Can you give you gnuradio script to plot that GSM vs PLL thing? Please 🥰🥰 There is also an #if 0 in you code, as I understand it was used to provide data to gnuradio?

mutability, 2020-08-25:
Uh, whatever script it was is >5 years old, I don't have any notes going back that far. Also no idea which `#if 0` you're talking about.

From distant memory it was just a shellscript and gnuplot wrapped around a modified version of `kalibrate`. You could do the same sort of thing with any combination of (stable frequency signal source) + (accurate frequency detector software). Repeatedly tune the detector to different offsets from the signal and measure the apparent baseband frequency.

If you wanted something easy to reproduce you'll probably want a signal generator.

erlendse, 2020-09-08:
For what it's worth or not, evrything inside the RTL-SDR run of a single clock and should be coherent,
so given R820T using the ring-osc in PLL(Phase Locked Loop) lock (non-sigma-delta) should give a clean crystal multiple.

So if it's recieved by the tuning in the tuner (sigma-delta PLL) with IF offset, and converted by the quadrature output DDC(Digital DownConverter) of the RTL, and the final frequency should be very predictable!
If you have control of the chain you would get zero frequency error from expected.

It's a good driver test, but unlikely to have a use for the end-users.

rrobinett, 2021-01-26:
I am trying to use the RTL-SDR in USB mode to receive WSPR beacon signals in the 200Hz wide band centered at 432.301500 Mhz.  The oscillator in the RTL-SDR V.3 is stable enough for that application and the pilot carriers of local ATSC TV stations provide an excellent off air calibration source, but like many others I find the unpredictable PLL tuning error makes it almost impossible to tune 1000 below the pilot carrier and get a 1000 hz audio tone out of rtl_fm.   
If someone wants to investigate this, I can share my bash script which logs the output audio tone as the tuning frequency changes in 1hz steps.