librtlsdr/librtlsdr #84: Next: implement dithering, register caching and batching and some thigs to rtl_power

ValZapod, 2020-08-24:
@hayguen I found a nice overview yesterday about keenerd tree on reddit https://www.reddit.com/r/RTLSDR/comments/2cix12/improvements_to_hopping_and_filters_testing/

BTW, please recheck that i2c repeater does what it was intended to do )) 

> Normally the i2c port is enabled and disabled between every single byte. Leave it open while its in use.

hayguen, 2020-08-25:
i'm not convinced, that this is enough .. but as keenerd writes at https://www.reddit.com/r/RTLSDR/comments/2cix12/improvements_to_hopping_and_filters_testing/ :

>  Possibly the secret sauce to phase aligned multiple dongles. I don't have the setup to operate this or the math to confirm. rtl_sdr -N or rtlsdr_set_dithering() to access it.

phase aligned and fully synchronized dongles are necessary for DF (direction finding)

ValZapod, 2020-08-25:
> phase aligned and fully synchronized

That is to complex stuff for me already I am afraid. Are we talking about phased array here?

ValZapod, 2020-08-25:
Anyway, that worked for KerberosSDR...

hayguen, 2020-08-25:
> Are we talking about phased array here?

don't think so, just direction finding and/or locating. but i'm not sure about the correct wording.
however, this stuff requires multiple dongles to be synchronized, e.g. at startup with a noise source.
think, https://www.rtl-sdr.com/ksdr/ is a good starting point.

IMHO, this dithering is not sufficient .. and there will remain a difference from the DDC (digital down converter)'s start time / sort of block processing. with enough filtering the difference might get small enough!? .. but i've never tried this kind of stuff ..

CornN64, 2020-08-25:
In order to keep the phase between a clock reference and a dongle (or two dongles assuming both are driven from the same clock source) constant without a slow drift due to the dithering you need to disable the dithering bit. This has the side effect that the center tuning will be in steps of ~439Hz but that is ok if you need absolute phase... 
 
#if 1
	//PLL dither OFF
	val |= 0x10;
#endif

	rc = r82xx_write_reg_mask(priv, 0x12, val, 0x18);

Some additional info can be found here (incl. proper sdm calculations...)
[http://michelebavaro.blogspot.com/2014/05/gnss-carrier-phase-rtlsdr-and.html](url)

Cheers