librtlsdr/librtlsdr #85: extend frequency range - with automatic sideband switching

hayguen, 2020-08-24:
explicit (manual) sideband switching was implemented (here) to achieve better filter quality/attenuation at the necessary sideband. see page 26 of https://codingspirit.de/librtlsdr-driver.pdf 

however, as seen in https://github.com/librtlsdr/librtlsdr/issues/57#issuecomment-678485965 , shifted frequency ranges can be achieved using different tuner sidebands. thus, when PLL fails to lock, the sideband could be switched internally / automatically

when i remind correctly, that was also the idea of https://github.com/mutability/librtlsdr

hayguen, 2020-09-19:
Implemented reception on fifth harmonic for frequencies above 1.76 GHz.
see https://github.com/librtlsdr/librtlsdr/issues/91#issuecomment-695214717

erlendse, 2020-10-13:
R18 (0x12), bit 0 seems to be a PLL analog charge pump range settings.

Setting it seems to give lock indication higher up in frequency (tested on airspy).