librtlsdr/librtlsdr #87: "Perf: Replace rtl_tcp linked list with ring buffer" for improved raspberry pi performance

CeruleanSky, 2020-08-25:
I may of overlooked it, but I didn't see it mentioned about a pull request from oscom's github mirror for a ring buffer for rtl_tcp, and thought it would be good to have it noted somewhere here due to how much is being improved in this fork.

https://github.com/osmocom/rtl-sdr/pull/6

It offers significant performance improvements even on a raspberry pi zero w allowing wifi usage:

>Removed the linked list structure, locking, and malloc/free on each USB callback.
Replaced it with a non blocking ring buffer for performance. This resulted
in a large performance improvement when using an Raspberry Pi or Raspberry Pi Zero W
as a rtl_tcp server. The sample rate could be doubled and instead of wired ethernet
wifi could be used. Frequnecy change lag was reduced greatly also.

hayguen, 2020-09-03:
with comments like

> Every one of these loads and stores of ringbuf_head and ringbuf_tail could race against loads and stores in another thread.

> I don't think completely replacing the linked list with a ringbuffer without still offering the linked list as an option is a good idea.

> The motivation behind this PR is actually specifically for raspberry pi performance

in https://github.com/osmocom/rtl-sdr/pull/6.

i confess, that i'm unsure if this patch should be in this general non-cpu specific repository.
other thoughts?

CeruleanSky, 2020-09-03:
My main goal was to help highlight other patches floating around that are easy to overlook, this one in particular as it seemed popular and useful as a rtl-sdr.com blog post was written about it.

But since you said you are unsure, I'll make an attempt at advocating for it, hopefully not too poorly, to try to help you decide.

You mentioned this is a non-cpu specific repository, but using linked list implicitly suggest that the code is for powerful cpu's with fast access to resources.  
Ring buffers extend things not only for the raspberry pi zero, but for all cpu's that have at least the minimal access to resources and execution speed  as the raspberry pi zero has.
Thus the suggestion to have this not as a replacement but modifying it to make it an option would make sense if you choose to add the usage ring buffers.

hayguen, 2020-09-03:
> the suggestion to have this not as a replacement but modifying it to make it an option would make sense

for making an option, the _user_ needs a decision base:
- when/which CPUs are safe to use ring buffers?
- which CPUs/architecures would be _unsafe_?

can someone answer these questions?
without answers on this .. most users will simply activate ring buffers for performance reasons .. what might lead to problems

CeruleanSky, 2020-09-03:
Perhaps we can look for a decider from what motivated ring buffers to begin with. 

According to the author's [blogpost](https://web.archive.org/web/20190918133349/http://www2.stesoft.com/blog/index.php/2019/02/10/software-defined-radio-with-raspberry-pi-part1/) it was rtl_tcp producing messages such as `ll+, now 153` so perhaps if the number exceeds a threshold a suggestion would be printed to enable a ringed buffer, lower the samples per second, or check the network.  

For the threshold, I am sure there is a objective number in which the issues with the linked list exceed the issues with ring buffer and vice versa, but in lieu of that perhaps when "ll+, now ###" where ### is over a hundred might be a good number.

Jasoroony, 2021-01-03:
I think the question should be "Do I want to assign [what little] memory I have available, for buffers, to avoid dropping data when the network gets bogged down?  (It might catch up soon, unless my kids are really hammering those games today.)"..

There are no CPUs that are unsafe with ring buffers.  That is, when one thread only updates the head, and the other only updates the tail.  What was presented as an example issue, was pointing out what the OP had done wrong.  That example would be appropriate for a thread that updates both the head and tail, which is done here.

From the blog post:

```
In the worst case, so much data is in the buffer that the USB worker thread runs into the “tail” and
cannot write any new data. I decided in this case to trim the buffer, and move the tail ahead.
This would result in the oldest data being lost.
```

What he should be doing is not writing the latest data into the buffer (latest gets dropped) which makes it safe by only having the head and tail updated by their respective threads.  The test for head == tail is fine, which he points out.