librtlsdr/librtlsdr #89: error building devlopment on mac os catalina

theresiasnow, 2020-08-30:
Current HEAD

❯```
 make -j4
Scanning dependencies of target convenience_static
Scanning dependencies of target rtlsdr_shared
[  3%] Building C object src/CMakeFiles/rtl_raw2wav.dir/rtl_raw2wav.c.o
[  9%] Built target rtl_wavestream
[ 11%] Building C object src/CMakeFiles/convenience_static.dir/convenience/rtl_convenience.c.o
[ 13%] Building C object src/CMakeFiles/rtlsdr_shared.dir/librtlsdr.c.o
[ 15%] Building C object src/CMakeFiles/rtlsdr_shared.dir/tuner_e4k.c.o
/Users/tess/Project/private/librtlsdr/src/rtl_raw2wav.c:224:21: error: no member
      named 'st_mtim' in 'struct stat'
                                fraction = attr.st_mtim.tv_nsec / 1E9;
                                           ~~~~ ^
1 error generated.
make[2]: *** [src/CMakeFiles/rtl_raw2wav.dir/rtl_raw2wav.c.o] Error 1
make[1]: *** [src/CMakeFiles/rtl_raw2wav.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 17%] Building C object src/CMakeFiles/convenience_static.dir/convenience/convenience
```

hayguen, 2020-08-30:
simply deactivated sub-second precision, which is not soooo important.
can you try again?

theresiasnow, 2020-08-30:
yes found it and fixed myself

```
diff --git a/src/rtl_raw2wav.c b/src/rtl_raw2wav.c
index 8e7d332..7865c7f 100644
--- a/src/rtl_raw2wav.c
+++ b/src/rtl_raw2wav.c
@@ -218,12 +218,21 @@ int main(int argc, char **argv)
 				gotmodtim = 1;
 			}
 #else
+	#ifdef __APPLE__
+			struct stat attr;
+			if (!stat(rawfilename, &attr)) {
+				tim = attr.st_mtimespec.tv_sec;
+				fraction = attr.st_mtimespec.tv_nsec / 1E9;
+				gotmodtim = 1;
+			}
+	#else
 			struct stat attr;
 			if (!stat(rawfilename, &attr)) {
 				tim = attr.st_mtime;
-				fraction = attr.st_mtim.tv_nsec / 1E9;
+				fraction = attr.st_mtime.tv_nsec / 1E9;
 				gotmodtim = 1;
 			}
+	#endif
 #endif
 			if (gotmodtim) {
 				long fs = 0;

```

theresiasnow, 2020-08-30:
your very quick fix works as well

hayguen, 2020-08-30:
looks, your patch doen't loose the fraction (sub-second precision).
you want to create a pull request? with the one liner  `fraction = attr.st_mtimespec.tv_nsec / 1E9;`

theresiasnow, 2020-08-30:
yes can do

theresiasnow, 2020-08-30:
done, working with bitbucket server daily so almost forgot the github PR way :-)

hayguen, 2020-08-30:
Everything else does now compile and even work now?

theresiasnow, 2020-09-03:
yes. it works. xcode with clang, have gcc also. 

```
❯ gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

```

hayguen, 2020-09-03:
great. thank you

closed 2020-09-03 by hayguen