librtlsdr/librtlsdr #91: heat issues

hayguen, 2020-09-03:
from @rtlsdrblog  by mail:

> Regarding temperature: I did spend quite a lot of time investigating the thermal performance issues with the RTL-SDR. Mainly to solve the issues regarding L-band PLL lock failure when the temp got too high as I wanted to make the dongles stable for use with 1.6 GHz satellites. So I just redesigned the PCB to allow for better heat spreading, and added the thermal pad interface to the metal case which solved it. There was also a register fix which provides more current to the PLL which helps a bit too. Not sure if you saw these register changes before? https://www.rtl-sdr.com/beta-testing-a-modified-rtl-sdr-driver-for-l-band-heat-issues/ Did you add those to the librtlsdr branch?

hayguen, 2020-09-09:
implemented options _vcocmin_/_vcocmax_ and _vcoalgo_ for R820T/2-tuners - as described in the help/usage or the commit message.
example calls:
```
rtl_tcp -a 192.168.224.128 -v -O vcocmin=7:vcocmax=7
rtl_test -t -f 1500M -e 2200M -O vcocmin=7:vcocmax=7:vcoalgo=0
rtl_test -t -f 1600M -e 2200M -O vcocmin=7:vcocmax=7:vcoalgo=2
```

i could not measure/notice a relevant change on temperature,
and also can't test real reception range without a signal generator going up to these frequencies.
i could only see/test, for which frequencies the PLL does lock.
for this and not breaking tuner precision with https://github.com/steve-m/librtlsdr/pull/10 i kept the default values - for now.

@rtlsdrblog or others: can you test?

erlendse, 2020-09-09:
Able to add lock detect polling? Just checking lock status without a retune that is.

The PLL may loose lock due to thermal reasons,
and it would give us a good idea about where to look next. (if someone can repeat the problem).

hayguen, 2020-09-10:
bugfix: now the options  _vcocmin_/_vcocmax_ and _vcoalgo_ for R820T/2-tuners are applied as described.

> i could not measure/notice a relevant change on temperature

this has to get tested again.

> Able to add lock detect polling?

added `rtlsdr_is_tuner_PLL_locked()` to the API and rtl_tcp is calling this every 3 seconds, when idle.
interesting: frequencies, e.g. ~ 1.88 GHz and several more report (PLL) error at `rtlsdr_set_center_freq()`, but report '_PLL locked_' later at `rtlsdr_is_tuner_PLL_locked()`

I have 2 DECT phones and can monitor the used frequencies with the DECT host (FritzBox) at around 1.88 GHz.
An RTL-SDR with E4000 tuner does show the DECT bursts in the waterfall/spectrum (of HDSDR through rtl_tcp-ExtIO). The R820T/2 tuner does report lock (not with rtlsdr_set_center_freq() but later), but just noise is visible.

hayguen, 2020-09-10:
fixed again: `rtlsdr_is_tuner_PLL_locked()` returned LOCK, when `previous rtlsdr_set_center_freq()` didn't set all relevant registers - or couldn't find appropriate settings for requested frequency.
that was the case for DECT 1.88 GHz :-(
added detailed error messages for verbose flag 'v', when setting frequency fails.

someone out there with a signal generator up to and over 2 GHz?

hayguen, 2020-09-19:
received some information from @rtlsdrblog via email concerning reception on **fifth harmonic** for L-band reception

hayguen, 2020-09-19:
Implemented reception on fifth harmonic for frequencies above 1.76 GHz.
Looks good in the Lab - without other signals around in the lower bands.
The signal level drop at least about ~ 16 dB when switching from straight tuning to the fifth harmonic.

Used the Adalm Pluto SDR as signal generator for those frequencies:
Could measure/see the generated carrier in the spectrum up to / over 2.2 GHz.
Holding a DECT phone near the RTLSDR, whilst it's input was connected to the PlutoSDR, showed up in the spectrum at ~ 1.88 GHz :-)

How it behaves in real world is an open question.
I'd assume aliases from lower frequencies - unless using an analog high pass filter in front of the RTL-SDR.

erlendse, 2020-09-19:
Try adjusting R23 (0x17) bit 1 & 2 (mask 0x06), should be LO power-level into the mixer. (unverified)

Changing the drive level is very likely to affect how the mixer deals with harmonics!

hayguen, 2020-09-19:
what to expect?
i can't see a difference on the generated carrier.

R23 bit 1 & 2 are still undocumented. you think, it's the power-level to mixer?
the default/initialization value for these is binary 00 - despite the documentation says bin 10.