librtlsdr/librtlsdr #92: add commands to rtl_udp

hayguen, 2020-09-06:
there were commands added to rtl_tcp. these are missing.

hayguen, 2020-09-21:
fixed with https://github.com/librtlsdr/librtlsdr/commit/40bef253739caa13bdf12572dd8bcbe96392011e

exception: rtl_udp is missing response channel

hayguen, 2020-09-21:
which application does use rtl_udp?
interesting to know for testing ..

erlendse, 2020-09-21:
For R820T, maybe add a command for returning R1 to R4 via UDP?

There is likely quite a bit of other get functions to implement.

Just don't allow it to reply to itself (target ip & port = own ip & port)

Get reply could possibly be similar to the set command?

hayguen, 2020-09-22:
had a look at the "response channel" in rtl_tcp:
* this is an additional TCP server, running on port = controlport+1 by default
* this "response" is (only) used to report all I2C registers every 500ms - after such a control command
* the name "response" indicates, that it was developed with more than I2C in mind
* TCP and an extra thread for the response channel would make synchronization necessary ..

with rtl_udp we could make similar logic .. with a simple response packet for every received/processed control command.
i could also add a GET_I2C_REGISTERS control command, specifying which registers (from - to) are requested

hayguen, 2020-10-05:
created new issue for 'response' channel: https://github.com/librtlsdr/librtlsdr/issues/98

closed 2020-10-05 by hayguen