librtlsdr/librtlsdr #96: Default parameters to match the older rtl_sdr

flux242, 2020-09-15:
Hello,

It seems like the default parameters are different to the older rtl_sdr. Somehow the gains are set differently I suppose. I'm not useing the new -O parameter but simply start:
```
rtl_sdr -s 2400000 -f 403000000 -g 40 -p $DONGLE_PPM
```

here how the signal power looks like with the old rtl_sdr (sample rate 2.4M): https://i.imgur.com/svNAmwv.png
here how the signal power looks like with the v. git0.8 (sample rate 2.4M): https://i.imgur.com/o5EIDPW.png
check the graphs at the lower right

The ends of the band are much more distorted and I'm using the whole 2.4MHz baseband

So, shouldn't the default parameters correspond to the older implementation? If not, then what should I specify with the -O to get the old behavior?

2. Did I understand correct that the samplerate up to 2.56M can be used with the new implementation? Because I tested it with rtl_test and I still can't go above 2.4M without loosing samples

Thank you

hayguen, 2020-09-15:
> The ends of the band are much more distorted and I'm using the whole 2.4MHz baseband

the ends don't look like distortion. you just see higher gain and the band filter edges.

> So, shouldn't the default parameters correspond to the older implementation?

ideally, yes. but, the problem arises cause the old/osmocom/librtlsdr (current master branch) are not consistent:

1. the RTL2832U chips IF control was deactivated all the time with
```
	/* disable RF and IF AGC loop */
	rtlsdr_demod_write_reg(dev, 1, 0x04, 0x00, 1);
```
2. initialization register contents in r82xx_init_array[] of tuner_r82xx.c was erroneously initialized: register 0x0c was initialized with value 0xf5, which means have the R820T/2's VGA (Variable Gain Amplifier) have controlled from RTL2832U's control line!
3. calling r82xx_set_gain() first time for automatic gain (set_manual_gain=0) does fix this, setting register 0x0c to the value 0x0b with mask 0x9f. this does deactivate automatic control of VGA through RTL2832U's control line - setting fixed gain.

hope you see the inconsistency.

meanwhile in this development branch, the RTL2832 does activate/use the control line .. and the tuner does (de)activate it's usage enabling higher gain (in total).
however fixing the previous bug at initialization state is IMHO sort of random - but reproducible (i think).
i'm not going to bring this _buggy_ behaviour back.

> If not, then what should I specify with the -O to get the old behavior?

for short, you can't get _exactly_ same behaviour as before.
but you might try to get similar behaviour with command line option between `-O ifm=10000` to `-O ifm=10011`.
from your screenshots we see ~ 20 dB more gain. looking at the gain table in tuner_r82xx.c:
```
#define VGA_BASE_GAIN	-47
static const int r82xx_vga_gain_steps[]  = {
	0, 26, 26, 30, 42, 35, 24, 13, 14, 32, 36, 34, 35, 37, 35, 36
/* -47, -21, 5, 35, 77, 112, 136, 149, 163, 195, 231, 265, 300, 337, 372, 408 */
};
```
you see 26.5 dB gain for the default index 0x0b = (decimal) 11.
20 dB less would correspond to +7.7 dB gain at index 4 at nearest.
(without) having measured .. `-O ifm=10004` or `-O ifm=77` should be best for previous behaviour.

alternative to the command line option, you might set the enviroment variable `LIBRTLSDR_OPT`:
```
export LIBRTLSDR_OPT="ifm=10004"
rtl_sdr -s 2400000 -f 403000000 -g 40 -p $DONGLE_PPM
```
or
```
LIBRTLSDR_OPT="ifm=10004" rtl_sdr -s 2400000 -f 403000000 -g 40 -p $DONGLE_PPM
```
as one-liner.

Please try and report your results!

> Did I understand correct that the samplerate up to 2.56M can be used with the new implementation?

No:
1. this was just experimental test at a few users. There no code-change for this.
2. below or equal 2.4M is OK. Just 2.56M did work for the few people who reported - but nothing between 2.4M and 2.56M!

hayguen, 2020-09-15:
oh my! sorry for all the stuff at begin. i didn't realize your `-g 40` option!
however the solution is quite similar, in the source, the default VGA value was different for AGC or manual gain:
AGC was VGA index 0x0b = (decimal) 11
With Manual gain the VGA index was 0x08 = (decimal) 8.
With explicit `ifm` option, these different defaults for VGA got removed. The new default is 0x0b.

thus, try `-O ifm=10008` or `-O ifm=163`

flux242, 2020-09-16:
I kept the -g 40 and added LIBRTLSDR_OPT="ifm=10008" and here is the result

old librtlsdr: https://i.imgur.com/nxm3cGX.png
new librtlsdr: https://i.imgur.com/LOYcFLr.png

The result look similar but I believe that the signal gain is around 2db less then with the older implementation. I can't say for sure because the signal is constantly changing so to make the fair measurement I'd need a constant reference signal. And of course the edges are distorted more.

I also played around the ifm option without specifying the -g option and the optimal value was 10006 with the same -2db effect comparing to the older implementation

hayguen, 2020-09-16:
with `ifm=10009` you would increase gain from 16.3 to 19.5 dB: about 3 dB.
but i don't understand the difference. could measure/test this evening ..

without specifying `-g`, you get automatic gain control for LNA and Mixer. in effect, you won't be able to compare signal levels! but this is one of the few reasons to use explicit gain control with `-g` and a non-zero value.

> And of course the edges are distorted more.

i wouldn't name that 'distortion'. however you see band edges you could configure and widen bandwidth per option. i see, the widest filter is 1950 kHz .. i could add some higher bandwidths - which weren't my focus.

hayguen, 2020-09-18:
tested/measured the gain with osmocom and current development branch using an Adalm Pluto-SDR as signal generator at 403 MHz - as in your example:
```rtl_sdr -s 2400000 -f 403000000 -g 40 -n 24000000 -O ifm=10009```
is nearest to the osmocom release. the output wave is approx 1 dB higher than with osmocom.

from the console output i see following message (amongst others) with the latest changes:
> Bandwidth set to automatic resulted in 2200000 Hz.

hayguen, 2020-09-19:
there was a minor change to reduce power consumption with https://github.com/librtlsdr/librtlsdr/commit/233d63436a1e52d7cf8d584125fcd1cc66ccf585
signal level dropped by ~ 0.5 dB

closed 2020-10-04 by hayguen