librtlsdr/librtlsdr #99: samp_rate != real_rate

crroush, 2020-10-19:
SDR info:
- RTL2832U OEM
- Realtek
- RTL2838UHIDIR
- R820T/2 Tuner

I request an input Sample Rate of 2000000 Hz
librtlsdr.c computes a sample rate of 2000000.052982. Outputting from line 2115 the following message "Exact sample rate is: 2000000.052982 Hz"

However `dev->rate = (uint32_t)real_rate;` Sets the rate back to my input rate due to the truncation. 
What I don't know is if the dev->rate should really be a double to show the fractional value or is this an error in computation of the "real_rate":
`rsamp_ratio = (dev->rtl_xtal * TWO_POW(22)) / samp_rate;`
`rsamp_ratio &= 0x0ffffffc;`
`real_rsamp_ratio = rsamp_ratio | ((rsamp_ratio & 0x08000000) << 1);`
`real_rate = (dev->rtl_xtal * TWO_POW(22)) / real_rsamp_ratio;`

hayguen, 2024-03-24:
this question is now waiting for s "small" while.
what is the background of this question?
usually most application can only process integer rates .. what usually is good enough - as the relative error is really small

crroush, 2024-03-29:
The issue here is was this an error in librtlsdr.c and it should have been an integer value, or is that fractional value actually what the hardware is doing (seems unlikely given the hardware used inside the rtlsdr's).