librtlsdr/librtlsdr #100: I have a little problem with dm=4

ulrichloose, 2020-11-14:
I have a little problem ...

export LIBRTLSDR_OPT = "dm = 4"

In gqrx I selected the rtl-stick and the terminal also shows that the variable $ LIBRTLSDR_OPT is available and is also being evaluated.

process options 'dm = 4' from environment 'LIBRTLSDR_OPT'
rtlsdr_set_opt_string (): parsed direct sampling mode 4 == 4: use Q below threshold frequency application of parsed option returned 0

The problem is that with the first change from below -> above and vice versa the following happens:

Disabled direct sampling mode
Enabled direct sampling mode, input 2
Allocating 15 (non-zero-copy) user-space buffers

and it remains with direct sampling. It is only when you change the set frequency by a few hearts that it is switched correctly. The same happens with the opposite sign. It is "switched over" in each case and then the old value is immediately set again.

Am I doing something wrong or is it a small mistake?

ulrichloose, 2020-11-15:
I am now a little closer to the matter ...

There are two places in librtlsdr.c where the center frequency is set.

int rtlsdr_set_center_freq (rtlsdr_dev_t * dev, uint32_t freq)

and

int rtlsdr_set_center_freq64 (rtlsdr_dev_t * dev, uint64_t freq)

In both routines there is a switchover to direct sampling

if (dev-> direct_sampling_mode> RTLSDR_DS_Q)
rtlsdr_update_ds (dev, freq);

If you switch on LOG_API_CALLS && LOG_API_SET_FREQ, the problem becomes clear:

...
Setting tuner bandwidth to 2e + 06 Hz.
LOG: rtlsdr_set_center_freq64 (freq 0.000000 MHz)
LOG: rtlsdr_set_center_freq (freq 126.955000 MHz)
LOG: rtlsdr_set_center_freq (freq 3.855000 MHz)
LOG: rtlsdr_set_center_freq64 (freq 123.000000 MHz)

The ..freq64 routine always lags one frequency change behind and since the sampling mode is checked or set in both routines, an error occurs the first time the frequency threshold is exceeded.

Maybe a professional could check this bug and turn it off? I am just a hobbyist with little experience and skill.

ulrichloose, 2020-11-15:
https://github.com/librtlsdr/librtlsdr/blob/710d578901a08b091748d4fd4471e3327a4d099e/src/librtlsdr.c#L2271

if commented out, the problem is gone...

hayguen, 2020-11-22:
thank you very much for pointing out the problem and especially the solution!
i'll try to test and integrate your patch soon - in the next days ..

ulrichloose, 2020-12-16:
https://github.com/librtlsdr/librtlsdr/blob/710d578901a08b091748d4fd4471e3327a4d099e/src/librtlsdr.c#L2328-L2333

seems the better position for a patch...

you have to include a `dev->freq = freq;` and the problem is gone, because the the frequency used at this position is not the actual dev->freq and that is the lag behind.

Line 2271 can be untouched and it is possible to add a call to `rtlsdr_set_tuner_bandwidth` before that line, because in automatic switching the original set bandwidth get lost....