librtlsdr/librtlsdr #101: rtl_sdr: symbol lookup error: rtl_sdr: undefined symbol: rtlsdr_set_opt_string

edegraaff, 2020-11-22:
Hi, i have a brand new, clean image of ubuntu workstation on raspberry pi 4.
I wanted to replace the current rtl_sdr for this one, since i want to make use of some exta options availible in this driver for my rtl-sdr v3.
On other pi's with a basic linux without gui raspbian i was able to compile this one with no issues but now... i have this problem, compiling without any issues but when i start the driver i get this error. 
Any clues? 

rtl_sdr: symbol lookup error: rtl_sdr: undefined symbol: rtlsdr_set_opt_string

regards
eelco

hayguen, 2020-11-22:
please check with `ldd` if the executable is linked against the correct/new library!
if necessary try with `LD_LIBRARY_PATH=<path> rtl_sdr`, replacing the `<path>`
also check `which rtl_sdr`

edegraaff, 2020-11-22:
> please check with `ldd` if the executable is linked against the correct/new library!
> if necessary try with `LD_LIBRARY_PATH=<path> rtl_sdr`, replacing the `<path>`
> also check `which rtl_sdr`

Seems this solves
LD_LIBRARY_PATH=/usr/local/lib
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH
sudo ldconfig

closed 2020-12-15 by hayguen