librtlsdr/librtlsdr #111: Modern cmake warns about cmake_minimum_required

johanhedin, 2021-05-28:
Modern cmake (like 3.19) warns that compatibility with cmake older than 2.8.12 will be dropped in the future.

cmake 2.8.12 was released in 2013 so it should be quite safe to update the minimum required version from 2.6 to 2.8.12 to avoid the warning.

superkuh, 2021-05-28:
Please don't disable builds on older distros preemptively. There are people like myself out there still compiling rtl-sdr on machines running Ubuntu 14.04 Extended Security Maintenance with cmake version 2.8.12.2 which is still getting security updates until 2024.

johanhedin, 2021-05-28:
Thank you for the comment.

cmake 2.8.12.2 is fine. The warning I am referring to talk about versions _older_ than 2.8.12. So if you are running 2.8.12.2 the suggested fix will work just fine.

I have a CentOS 7 box myself and the stock cmake there is also 2.8.12.2 and my suggested fix work.

johanhedin, 2021-05-29:
The cmake discussion forum suggest another way to indicate to cmake that the configuration works fine on newer versions without affecting older ones:

https://discourse.cmake.org/t/how-to-fix-cmake-minimum-required-deprecation-warning/2487

So, in this case it will be:

``cmake_minimum_required(VERSION 2.6...3.19)``

johanhedin, 2024-01-11:
Fixed in #112.

closed 2024-01-11 by johanhedin