librtlsdr/librtlsdr #116: RTL SDR not getting powered down when rtlsdr_close is not called e.g. when process ends

nanoant, 2021-08-16:
It seems that when the application using `librtlsdr` exits without calling `rtlsdr_close`, then the dongle remains powered and hot even it is no longer used by any application. I can guess that maybe ADC and demodulators are not powered down.

Is there any way to enforce powering down or resetting the device when process using it exits? I know that apps should normally call `rtlsdr_close` on exit, but I can imagine it can very likely happen that process crashes or gets killed and we are left out with powered dongle.

I can also guess the problem is that librtlsdr uses driverless approach via libusb. And if we had a regular driver instead, then OS would normally run driver deinitialization on process exit.

FYI I am running RTL SDR Blog v3 on Intel macOS 10.15.7 with SDR++ app.

This is related to the issue reported here: https://github.com/AlexandreRouma/SDRPlusPlus/issues/261

hayguen, 2024-03-20:
i remind, there was some work on this .. can you check again?

hayguen, 2024-03-22:
that work was on my personal dev branch .. now moved that into `development` branch here.
but, that was an option, to keep the everything "ON" with `rtlsdr_close()`.
thus, it's the opposite of your wish

on a crash, nothing can be done! you might call `rtl_sdr -n 1024` to open and close device afterwards.

without a crash and a missing `rtl_sdr_close()` the device will be warm at least until application exits. in this case, some `at_exit()` code could be executed to get the rtlsdr_close() done.  but, in this case, you could also use the workaround as in the crash case.

so, i tend to ignore this issue