librtlsdr/librtlsdr #121: Issue with buffer lifetimes

zuckschwerdt, 2022-01-11:
I'd like to propose an addition to the async API.
Nothing is broken or unusable, this is purely a request to improve the API.

A reasonable or at least expected use case is to send the data from the `rtlsdr_read_async()` callback
to processing threads.
The buffers can disapear unexpectedly on errors or when `rtlsdr_cancel_async()` is called though.
We currently need to either block the callback until processing is finished or pessimistically `memcpy()` the buffers.

I'd like the contract to keep the buffers around until some processing and cleanup is finished.

I see two approaches:
- Introduce (opt-in) `rtlsdr_init_async()` and `rtlsdr_free_async()` to let the user choose when it's safe to remove the buffers.
  A draft PR is in #122
- Introduce (optional) `rtlsdr_cleanup_async()` and by default keep buffers until `rtlsdr_close()`.
  A draft PR is in #123

I slightly prefer the second approach, the logic is easier to follow with the only drawback that
some memory is kept around until `rtlsdr_cleanup_async()`, `rtlsdr_close()` or the next `rtlsdr_read_async()`.

Is this a reasonable idea and ok to extend the API and contract? Or would we recommend to just `memcpy()` received data as we don't want the extra book-keeping? And am I perhaps missing some corner case where the buffers really need to disapear quickly?

zuckschwerdt, 2022-01-11:
The first approach (optional init/free for buffers) could also implement the proposed use-case from #37, i.e. by default use the current logic but allow for either managed lifetime or fully external buffers.

zuckschwerdt, 2022-01-11:
Sorry, I might have made this sound more dramatic than it actually is. It's purely about making life easier (and more efficient) for the user of the lib.

For a use case here assume something like a rtl_tcp server clone:
- a thread `accept()`ing client connections and spawning a thread per client
- a loop blocked on `rtlsdr_read_async()`

The user-callback from `rtlsdr_read_async()` now needs to `memcpy()` all received data to owned buffers before handing off to the threads. The reason being that any exit from rtlsdr_read_async() will `free()` the buffers and a thread will likely be blocked in `send()`. That use-after-free can be prevented if the buffers stay around a little longer so we can cleanly shut down.
That's all assuming a worst-case where we'd like zero-copy, need concurrency or responsiveness and are not satisfied with just crashing on exit. (The example could be done with an event loop, or just serve the connection in the callback, or just accept that an unplanned exit from `rtlsdr_read_async()` is an error anyway and crash with segmentation violation is just as well).

It all works well as-is, I just copy the buffers handed in by the callback to my own buffers -- it just feels like there is an opportunity to do better ;)

hayguen, 2022-01-11:
now, i see the point:
you would like to process the buffer arriving at the callback - directly, without prior copying the buffer contents to a safe place.

in this case, i have following questions:
1- would you process the data directly in the callback? or would use another thread anyway?
2- how to detect a buffer overrun? how does libusb act, when the buffer, it would like to write new data, is still busy with the callback?

by copying the data into own user buffers for a processing thread "quickly", it's possible to detect that situation.
see https://github.com/hayguen/librtlsdr/blob/development/src/rtl_multichannel.c#L563-L583

IMHO it's more important to detect an overrun, than squeezing a below few percents of performance.

for signal processing - as librtlsdr usually serves for, memcpy() is far cheaper than any "real" signal processing ..
however, if there's a way to detect an overrun, one of your PRs would make sense.

zuckschwerdt, 2022-01-12:
Indeed this falls squarely into the "measure then optimize" category, so no rush, a thorough discussion is needed.

To maybe summarize the observations: *It is currently not safe to keep buffer pointers beyond the end of any `rtlsdr_read_async()` user-callback. It would be nice-to-have if we could relax that, given some other condition are accounted for.*

re 1: I see uses for both cases. The callback could keep a limited number of old buffers to e.g. "pre-record" for an event trigger (e.g. an event is detected in the current buffer then write out the previous buffer too). But this is currently not a problem at all: if you re-enter the callback all old buffers are still there.
Handing the buffer to threads I wouldn't want the thread to lag a (variable) number buffers behind, but rather to gurantee the current buffer no matter the order of eventual shutdown.

re 2: I assume the user is not writing to the buffer, a [`libusb_dev_mem_alloc()`](https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#ga47c492e750197f1a28874c7a0dee40e3) might be used, and as stated it's not safe to modify the buffer at all. The data is valid to read anytime, though it might be garbage on overrun. We would track the number of callbacks since getting the buffer and e.g. a buffer "N callbacks old" indicates an overrun.

In my case (in threads) I immediatly drop old buffers on getting a new one though. Your example is a ring buffer, right? I used a circular buffer just now in a very pragmatic way: Usually you need a mutex to relate tail and head pointer and act on overrun. Realizing that an overrun condition here has just one choice, skip the head further along, I could just use independent atomics for head and tail, they don't interfere in writes anymore (moving the tail beyond the head notes an overrun and the head has now skipped an entire buffer implicitly without the need for write-access) -- that however assumes that I don't care if slow consumers get garbage data ;)

Detecting overruns in user code would be a scheme of:
- if you keep old buffer pointers
- you need to known the number of buffers
- and take overrun action if one nears or exceeds the other.

Also one of the following needs to be observed:
- either be prepared to read garbage from old pointers
- or ensure that old pointers are dropped quickly

The later one is my use-case, e.g. at 1MS/s I have leisurely 100ms to update every processing thread. (a ring of just 3 buffers could safely work: old, new, and active usb transfer).

zuckschwerdt, 2022-01-13:
I did some futher exploring with buffers. E.g. to copy data out MSG_ZEROCOPY is disappointing, in most cases it signals that a copy was still necessary. Not really worth to bother with that API.
I will just use `sendfile()` for network and file output. That mandates mmap memory. I'll just copy all received frames to my mmap memory in the callback and that covers my use-case. It's efficient because: I know there is USB DMA/zerocopy, so putting the data anywhere else is a memcpy anyway and I can do that as user.

This leaves the use-case of #37 where buffers need to be user controlled, and a possible chance to optimize where the user knows there is no USB zerocopy and wants the first copy to go somewhere specific, like mmap memory.

If there is to be any API addition I would now propose a `int rtlsdr_setup_async(rtlsdr_dev_t *dev, uint32_t buf_num, uint32_t buf_len, unsigned char **bufs)` to (optionally, opt-in) setup user controlled buffers for use in async read or clear them (using `NULL` as arg).
I'll add a draft PR to showcase this.

zuckschwerdt, 2022-01-31:
Closing this for now, the only substantial benefit is for situations like #37.

closed 2022-01-31 by zuckschwerdt