librtlsdr/librtlsdr #125: rtl_fm: undefined symbol: rtlsdr_set_ds_mode

kovalroma, 2022-03-10:
```
rtl_fm -f 144M:146M:12.5k -M nfm -s 12.5k -l 32 -t 1 -g 30
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Tuner gain set to 29.70 dB.
rtl_fm: symbol lookup error: rtl_fm: undefined symbol: rtlsdr_set_ds_mode
```
```
uname -a
Linux raspberrypi 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux
```

```
 rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 196 bytes in buffer 0
```

Any idea how to fix ?

hayguen, 2022-03-10:
looks, the system search path "/usr/lib/arm-linux-gnueabihf/" with system-wide installed librtlsdr is in the way - despite the binary is taken from the freshly compiled one at "/usr/local/bin".
you might check with "which rtl_fm" and "ldd rtl_fm" for the used libraries.

you might
* uninstall system installed librtlsdr
* recompile/install the librtlsdr tools - statically linked against the library (cmake option LINK_RTLTOOLS_AGAINST_STATIC_LIB )
* prefix rtl_fm call with LD_LIBRARY_PATH="/usr/local/lib/"

kovalroma, 2022-03-11:
@hayguen thank you so much !

This solve my problem. 
```
/librtlsdr-0.8.0/build $ sudo make uninstall
cmake ../ -DINSTALL_UDEV_RULES=ON -DDETACH_KERNEL_DRIVER=ON -DLINK_RTLTOOLS_AGAINST_STATIC_LIB=ON
make
sudo make install
sudo ldconfig
```

closed 2022-03-11 by kovalroma