librtlsdr/librtlsdr #127: rtlsdr_demod_write_reg failed with -9 on new PC

dvdesolve, 2022-05-17:
I'm on Arch Linux 64 bit, latest updates installed. Software versions:
- gqrx 2.14.4
- gnuradio 3.8.2.0
- gnuradio-osmosdr 0.2.3
- rtl-sdr 0.8.0

I've migrated this install from Asus VivoMini VC65 / Intel i3-6100T setup to the Asus PRIME B550-PLUS / Ryzen 3900XT. I've noticed that when using RTLSDR v3 dongle and setting reception frequency in GQRX the following errors spams console:
```
...
rtlsdr_demod_write_reg failed with -9
r82xx_write_arr: i2c wr failed=-9 reg=17 len=1
...
```
and so on. As a result GQRX doesn't set frequency correctly, only on occasional success. I've even filed a bug (https://github.com/csete/gqrx/issues/953) on their GitHub but then I've realized that the problem is with rtlsdr library itself. I've tried different RTL SDR dongles (V3 and versions from Aliexpress) but with the same result. I've tried to connect dongle directly to the USB ports, have tried different ports (2.0 and 3.0, back and front) - but with no positive result. I've even tried to update UEFI to a several new versions and then mangle with UEFI USB settings such as IOMMU, XHCI hand-off and so on - and with no luck. At the same time Airspy R2 works just well with this setup so it seems like a problem related to the way how librtlsdr communicates with RTLSDR dongles.

It would be very helpful to get some advice.

alxpettit, 2023-04-04:
I'm getting this same bug on NixOS 22.11. :/

dvdesolve, 2023-04-05:
Can you please provide some info about your hardware?

alxpettit, 2023-04-06:
The dongle reads `RTL2832U R820T2 TCXO+BIAS T+HF`.

alxpettit, 2023-04-06:
Looks like it's just very sensitive to the USB port it's plugged into. I found a workaround by plugging it into an external USB hub.

mkalte666, 2023-04-19:
This is probably a power issue, as the v3 dongle draws quite a bit. I have made myself a power adapter that inserts an external usb power supply before, and the dongle is running fine now without the need of a usb hub. 

If you intend to build such an adapter yourself, think about your power situation a bit. The suppy you use should be isolated so you can safely connect the grounds together without running into issues with your computers PSU.

dvdesolve, 2023-04-21:
cheap Chinese blue dongle from famous marketplace (not V3 version) behaves the same way
and it's happening on the system with quite heavy duty PSU and MB
even on simple, old and weak laptop it works fine
I still think that the problem may be related to the USB driver implementation in linux kernel

atomikdg, 2023-11-20:
Same problem here with Gqrx 2.17.3, kernel 6.6.1 and a nooelec nesdr smart.
@dvdesolve  did you find a solution ?

Cherenkov11, 2023-11-26:
@atomikdg I encountered with this problem by accident. I used to had my rtl-sdr v3 connected to one of my USB3 ports of my pc (AMD B650) in it worked perfectly fine. But recently I redid my cable management and I connected my sdr to one of the USB2 ports and this issue of "rtlsdr_demod_write_reg failed with -9" started to appearing. So, in summary; on my end if I have my sdr connected to my USB3 ports (excepted the one on the front) everything works fine, no "rtlsdr_demod_write_reg failed with -9" but any USB2 port or the front ports of my PC throws the error.

One thing to notice, I use linux and for a moment I thought that this was a linux-issue, but no, even on windows rtl_test throws the same error on any USB2 port. 

I'm not completely sure, but I thing this is incompatibility (maybe race condition) between librtlsdr and the USB2 controller present on AMD chipsets. Because on my Intel based machine all the USB2 ports work perfectly fine. Maybe a developer from rtlsdr could guide us to provide more debug information to fix this issue.

dvdesolve, 2023-12-05:
I still haven't find a solution. iirc, I've tried all of the available USB ports. I've even tried to play with USB settings in uefi but with no luck
thank you for pointing out towards rear usb3 ports - I'll give it a try once again

dvdesolve, 2024-01-06:
Just tested on the desktop from the first post. @Cherenkov11 , you were right. When I plug in the dongle to the rear usb3 port directly or via extension cable, all works fine.
When I use front usb2 port I can't even start listening in GQRX with the following errors:
```
...
Allocating 15 zero-copy buffers
cb transfer status: 1, canceling...
rtlsdr_read_async returned with -5
Allocating 15 zero-copy buffers
Failed to allocate zero-copy buffer for transfer 0
Falling back to buffers in userspace
Allocating 15 (non-zero-copy) user-space buffers
Failed to submit transfer 0
Please increase your allowed usbfs buffer size with the following command:
echo 0 > /sys/module/usbcore/parameters/usbfs_memory_mb
rtlsdr_read_async returned with -5
...
```
Using the dongle in front usb3 port gives famous errors like below:
```
...
Allocating 15 zero-copy buffers
rtlsdr_demod_write_reg failed with -9
r82xx_write_arr: i2c wr failed=-9 reg=17 len=1
rtlsdr_demod_write_reg failed with -9
r82xx_write_arr: i2c wr failed=-9 reg=17 len=1
rtlsdr_demod_write_reg failed with -9
r82xx_write_arr: i2c wr failed=-9 reg=17 len=1
rtlsdr_demod_write_reg failed with -9
r82xx_write_arr: i2c wr failed=-9 reg=06 len=1
...
```

atomikdg, 2024-01-18:
@Cherenkov11 No USB from my motherboard (asus prime b450-plus) is working with the dongle/driver. I tried all of them.
I've bought on amazon a Inateck RedComets U25 controller and it works perfectly.

Thanks for all the informations

hayguen, 2024-03-21:
are you definitely using this fork?
manually compiled?
or some osmocom variant from your distri?
would you try manual compilation with the current development branch?

atomikdg, 2024-03-21:
I've bought an Inateck RedComets U25 and now there is NO problem with it. It works perfectly ! So it was the AMD chipset or something hardware related.

closed 2024-03-22 by hayguen

cybekRT, 2025-01-31:
I've seen this issue from: https://github.com/gqrx-sdr/gqrx/issues/953
Even if the problem is USB-related, shouldn't the UI be aware of the error and do not change the values displayed if the frequency wasn't properly changed?