librtlsdr/librtlsdr #128: Memory leak in rtl_power.c

SamG-git, 2022-06-19:
Valgrind detects a memory leak when running rtl_power.c on like 810. `freq_optarg` is allocated on line 810 with 'strdup' but never freed. Adding `free(freq_optarg)` before returning from `main` fixes this.

zuckschwerdt, 2023-10-03:
There is no leak from one-off allocations, exiting will return the memory. But it might be better to do that so memory checkers have relevant output.

Just adding the `free()` as suggested is a bug as `freq_optarg = "";` is initializes with a static string.
The `strdup()` could be removed from the assignment though. The memory from argv will be around (but changed by `frequency_range()`).

Otherwise the init needs to be to `NULL`, before the `strdup()` needs to be a `free()`, then one somewhere after `frequency_range()` and a `NULL` check in that function.

Overall no need to touch this code though.

hayguen, 2024-03-22:
fix is in release v0.9

closed 2024-03-22 by hayguen

zuckschwerdt, 2024-03-22:
That fix has some holes. E.g. `-f 42 -f 42` will leak the first `strdup()` and `-f ""` will leak the empty `strdup()`.

Do you want a PR for a proper fix?