librtlsdr/librtlsdr #132: Problems in ubuntu 22.04 (Kernel 5.19.0)

ariyan-eghbal, 2023-07-13:
I am trying to use `librtlsdr` in ubuntu 22.04. 
I compiled and installed it as README file with this `cmake` arguemts:  

`-DINSTALL_UDEV_RULES=ON -DDETACH_KERNEL_DRIVER=ON -DPROVIDE_UDP_SERVER=ON -DWITH_RPC=ON -Wno-dev`  

But when trying `rtl_test -t` it starts and finds device without problem but exists without any error (exit code 0):
```
$ rtl_test -t
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U
Found Rafael Micro R820T/2 tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
Sampling at 2048000 S/s.
Testing tuner range. This may take a couple of minutes..
$ 
```

I also tried to test with `GQRX` and `SDR++`.
`GQRX` doesn't list device at all but there is some stranger things with `SDR++` , it detects device and some features like `Bias-T` works! but it only shows white noise without any signal !

The `Bias-T` feature also works with `rtl_biast` (LED on LANA turns on) 

My device is `RTL2838UHIDIR` with vid/pid of `1209/2832` and kernel is not loading any dvb drivers for it (modules are also blacklisted):

```
$ dmesg
[ 2285.978010] usb 3-3: new high-speed USB device number 11 using xhci_hcd
[ 2286.143504] usb 3-3: New USB device found, idVendor=1209, idProduct=2832, bcdDevice= 1.00
[ 2286.143514] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2286.143518] usb 3-3: Product: RTL2838UHIDIR
[ 2286.143520] usb 3-3: Manufacturer: Realtek
[ 2286.143523] usb 3-3: SerialNumber: 00000001

$ lsmod | |grep -iE 'dvb|rtl|820'
$
```

I tried to debug `udev` and it seems it is using `/etc/udev/rules.d/rtl-sdr.rules` file (full udev debug report is available at the end of issue):

`Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: /etc/udev/rules.d/rtl-sdr.rules:22 MODE 0666`

I also tried to uninstall `librtlsdr` and test the old steve-m's one from ubuntu's official repository. strange thing is the original module cannot detect device at all and reports `No supported devices found.` !!

I tried my dongle on my old system with ubuntu 20.04 and everything works well !

Any idea what is the problem and why module is not working? 

full udev debug output after plugging dongle is:

```
Jul 13 14:10:40 ryn-ThinkPad kernel: usb 3-3: new high-speed USB device number 10 using xhci_hcd
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[711]: 3-3: Device is queued (SEQNUM=6460, ACTION=add)
Jul 13 14:10:40 ryn-ThinkPad kernel: usb 3-3: New USB device found, idVendor=1209, idProduct=2832, bcdDevice= 1.00
Jul 13 14:10:40 ryn-ThinkPad kernel: usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jul 13 14:10:40 ryn-ThinkPad kernel: usb 3-3: Product: RTL2838UHIDIR
Jul 13 14:10:40 ryn-ThinkPad kernel: usb 3-3: Manufacturer: Realtek
Jul 13 14:10:40 ryn-ThinkPad kernel: usb 3-3: SerialNumber: 00000001
Jul 13 14:10:40 ryn-ThinkPad mtp-probe[11632]: checking bus 3, device 10: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-3"
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[711]: Validate module index
Jul 13 14:10:40 ryn-ThinkPad mtp-probe[11632]: bus: 3, device: 10 was not an MTP device
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[711]: Check if link configuration needs reloading.
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[711]: 3-3: Device ready for processing (SEQNUM=6460, ACTION=add)
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[711]: Successfully forked off 'n/a' as PID 11631.
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[711]: 3-3: Worker [11631] is forked for processing SEQNUM=6460.
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11631]: 3-3: Processing device (SEQNUM=6460, ACTION=add)
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11631]: 3-3: Failed to get watch handle, ignoring: No such file or directory
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[711]: 3-3:1.0: Device is queued (SEQNUM=6461, ACTION=add)
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[711]: 3-3:1.0: SEQNUM=6461 blocked by SEQNUM=6460
[... 61 lines cut ...]
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: Failed to find module 'usb:v1209p2832d0100dc00dsc00dp00icFFiscFFipFFin01'
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3:1.1: Device processed (SEQNUM=6462, ACTION=add)
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3:1.1: sd-device-monitor: Passed 409 byte to netlink monitor
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[711]: 3-3: Device ready for processing (SEQNUM=6463, ACTION=bind)
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[711]: 3-3: sd-device-monitor: Passed 290 byte to netlink monitor
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: Processing device (SEQNUM=6463, ACTION=bind)
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: Failed to get watch handle, ignoring: No such file or directory
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: value '[dmi/id]sys_vendor' is 'LENOVO'
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: value '[dmi/id]sys_vendor' is 'LENOVO'
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: /usr/lib/udev/rules.d/50-udev-default.rules:13 Importing properties from results of builtin command 'usb_id'
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: /usr/lib/udev/rules.d/50-udev-default.rules:13 Importing properties from results of builtin command 'hwdb --subsystem=usb'
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: hwdb modalias key: "usb:v1209p2832:RTL2838UHIDIR"
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: /usr/lib/udev/rules.d/60-drm.rules:3 Importing properties from results of builtin command 'path_id'
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: /usr/lib/udev/rules.d/69-libmtp.rules:39 Running PROGRAM 'mtp-probe /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3 3 10'
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: Starting 'mtp-probe /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3 3 10'
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: Successfully forked off '(spawn)' as PID 11634.
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: 'mtp-probe /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3 3 10'(out) '0'
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: Process 'mtp-probe /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3 3 10' succeeded.
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: /etc/udev/rules.d/rtl-sdr.rules:22 MODE 0666
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: Handling device node '/dev/bus/usb/003/010', devnum=c189:265
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: sd-device: Created db file '/run/udev/data/c189:265' for '/devices/pci0000:00/0000:00:14.0/usb3/3-3'
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: Failed to get watch handle, ignoring: No such file or directory
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: Device processed (SEQNUM=6463, ACTION=bind)
Jul 13 14:10:40 ryn-ThinkPad systemd-udevd[11633]: 3-3: sd-device-monitor: Passed 711 byte to netlink monitor
```

hayguen, 2024-03-22:
noticing your issue. do you still have this issue?
have tried to compile a debug version and start rtl_test with gdb?
but you even don't get a "core dump" message!?

kenwawa01, 2024-06-24:
Please have a look on this thread: https://github.com/merbanan/rtl_433/discussions/2968

However it looks like that after writing that everything works ok, I discovered that... it's not always the case and it sometimes happens that I'm getting error codes but much more seldom. If it starts working, it works without problems. But sometimes I have to "kick the engine" few times...