librtlsdr/librtlsdr #134: Release new version

AsciiWolf, 2023-09-18:
Last release was in 2020. There were many changes, fixes and updates (including the new RTL-SDR.com V4 dongle support) since then. Please, consider making a new release of librtlsdr.

/cc @hayguen

gdt, 2023-10-31:
Releases are critical in the packaging world.  They are how a project announces that a particular commit is stable and should be used.   I'm writing as someone who works on pkgsrc and one of the people that works on rtl_433.

As part of a release, please also adjust the README to clarify the relationship to the osmocom repo.   Is this a temporary maintenance fork, and you expect your changes to get merged there?  Is it  a continuation fork, basically deciding the osmocom is no longer maintained, and therefore you suggest packagers to change to you as upstream?  Is it something else?

kerel-fs, 2023-11-02:
Only two weeks ago rtl-sdr was [added](https://gitea.osmocom.org/osmocom/osmo-ci/commit/7fbb49f87d2ab4f320fe4896e04c2b79e9aa508b) to the [osmo-ci](https://gitea.osmocom.org/osmocom/osmo-ci) continuous integration pipeline, such that automated release tarballs are available now: https://ftp.osmocom.org/releases/rtl-sdr/

Upstream is accepting & merging patches, e.g. the rtl-sdr v4 patch can be found in the master branch. That said, there has not been a new upstream release (yet?) since 0.6.0 (2018-08-26).

testuser7, 2023-12-03:
New version `v2.0.1` is released: https://github.com/steve-m/librtlsdr/releases/tag/v2.0.1

wooferguy, 2024-03-16:
Bumping this. There are other projects that rely on this repo to fetch the latest version of the library from the releases, namely rtl_433 and pyrtlsdr. Rather than pointing to source code that needs to be compiled on a specific environment, please add a pre built release.

hayguen, 2024-03-16:
sorry guys, just realizing this issue.
any volunteers to work on this repo .. and release new work?
in had some some stuff in my personal fork/work, but i currently don't know of the status there .. had been unable to catch up on librtlsdr

AsciiWolf, 2024-03-17:
For now, just bumping the version number and making a new Git tag should be enough. The rest can be solved later.

hayguen, 2024-03-17:
what version should it be?

AsciiWolf, 2024-03-17:
The version number should probably be the same as the steve's (or osmocom) one: https://github.com/steve-m/librtlsdr/tags

So, 2.0.0? (2.0.1 was just some bugfix not relevant to your version.)

hayguen, 2024-03-17:
can someone check, that we have at least all api functions, steve's or osmocom's 2.0 has?
for the amendments, i would then say, to have 2.1.0, if we have ALL api functions. agree?

hayguen, 2024-03-20:
just compared the include header: all api functions of
https://github.com/steve-m/librtlsdr/tree/v2.0.1
did already exist. fine.

this fork provides a bunch of additional api functions,
thus, i set the library's version to 2.1:
https://github.com/librtlsdr/librtlsdr/blob/v2.0.1/CMakeLists.txt#L51-L52

and created the tags
https://github.com/librtlsdr/librtlsdr/tree/v2.0.1  and
https://github.com/librtlsdr/librtlsdr/tree/v2.1.0

all fine?

wooferguy, 2024-03-21:
In previous releases there have been builds included, any chance of those being added too?

hayguen, 2024-03-21:
put the files from automatic github actions build to the release.
there was also a travis build mechanism, but i never had used that. don't know.

testuser7, 2024-03-21:
@hayguen Do you have plan to synchronize with https://github.com/steve-m/librtlsdr repo?
This commit may be helpful in some cases: https://github.com/librtlsdr/librtlsdr/commit/142325a93c6ad70f851f43434acfdf75e12dfe03

hayguen, 2024-03-21:
> @hayguen Do you have plan to synchronize with https://github.com/steve-m/librtlsdr repo? This commit may be helpful in some cases: [142325a](https://github.com/librtlsdr/librtlsdr/commit/142325a93c6ad70f851f43434acfdf75e12dfe03)

sync: only on request for interesting featues.
please create a new issue for that commit

rtlsdrblog, 2024-03-22:
Hey, so the major version number bump is going to cause some big software incompatibility problems. When the major version is incremented it makes a new soname, so this means that all previously installed SDR programs that make use of librtlsdr will need to be recompiled from scratch.

Also if people don't properly uninstall the older version, and install the newer version, it's going to look like the driver update didn't work to most programs, until they are recompiled. Because the software will continue using the older driver version. Big headache for people trying to add Blog V4 support.

And I believe that software from the package manager that's looking for the older soname simply won't work anymore if the newer 2.0 version is installed.

The major version change is not even needed as there was no breaking API changes. It's going to be a huge headache for several years to come if it's not reverted IMO.

We're already seeing problems with this change (from the Osmo repo) causing big support issues for software like SDR++.

rtlsdrblog, 2024-03-22:
Here is a Twitter/X thread from the dev of SDR++ who probably explains it better: https://twitter.com/ryzerth/status/1771016439681466697

hayguen, 2024-03-22:
Thank you.
Thus, i will revert the increase of the major version number ..
and create a new release tag: 0.9

hayguen, 2024-03-22:
v0.9 is now ready to go:
https://github.com/librtlsdr/librtlsdr/releases/tag/v0.9.0

unsure, if i should completely remove release 2.1
https://github.com/librtlsdr/librtlsdr/releases/tag/v2.1.0
together with its release tags?

rtlsdrblog, 2024-03-22:
I'm not sure about the version numbering in the long run. 

I assume that if Osmo doesn't revert too (unlikely since some distros already picked up librtlsdr2), we'll all eventually need to move to v2.0 releases once software devs start creating packages and releases that look for librtlsdr2, and once distros start packaging librtlsdr2.

hayguen, 2024-03-22:
I could imagine to create a copy or symlink to both versions when building..